sinara-hw/Phaser #1: design sketch

hartytp, 2019-10-18:
2 channel ADC + DAC board

Hardware design sketch, needs fleshing out...

Digital:
- 2 EEM connectors
- TBD FPGA (probably small/medium Artix 7)
- Same RAM as Kasli
- Maybe a SATA or two
- Possibly, but probably not, Si5324 for DRTIO (would need to check power budget carefully to make sure this feasible)

Analog Input:
- Same as Sampler except
- Go for the 5MSPS ADC in the same family
- Increase AFE pole frequencies to ~1MHz to give more bandwidth

DAC:
- TBD DACs
- Two AFE population variants

Baseband AFE:
- Only using DAC I output, Ignore (terminate?) DAC Q outputs
- same AFE as Urukul
- Probably drop the pre-amps to keep the power dissipation down (in any case, with Booster we have gain to spare)
- Optionally drop the switches too

Upconversion AFE:
- ADL5375 IQ mixer (I and Q channels)
- same AFE as baseband mode, but different (no?) filter and magnetics

Clocking:
- 125MHz from SMA/MMCX/on-board XO
- HMC830 DAC clock PLL (copy loop filter from Sayma)
- HMC830 LO clock PLL (copy loop filter from Sayma)
- Decent LVDS clock fanout buffer

gkasprow, 2019-10-19:
Will we use 2x LTC2000 DAC then mixer and a single output?
Or two dual DACs and two mixers?

What about making the design 3-channel. Two baseband channels and one with upconversion? In this way switching from one mode to another won't require that many changes.

gkasprow, 2019-10-19:
do you want USB JTAG or we can use solution from Banker that Kasli can upgrade config FLASH via I2C ?

hartytp, 2019-10-19:
>Will we use 2x LTC2000 DAC then mixer and a single output?

The DAC will most likely be http://www.ti.com/lit/ds/symlink/dac34sh84.pdf

This has two IQ pairs. So, the idea is we either use only the Is and drop the Qs (baseband) or use an external IQ mixer depending on the desired output frequency range.

> What about making the design 3-channel. Two baseband channels and one with upconversion? In this way switching from one mode to another won't require that many changes.

We can't make three channels work in terms of thermal/power management, (front panel) space, EEM link bandwidth.

The proposal would be to just have a population option to select (basically, just a solder jumper to bypass the mixer). If that turns out to be hard to do, we can scrap the upconversion *the key element of this design is baseband performance, upconversion is an optional extra*.

hartytp, 2019-10-19:
> do you want USB JTAG or we can use solution from Banker that Kasli can upgrade config FLASH via I2C ?

I'll leave all those kinds of decisions to @jordens

jordens, 2019-10-20:
Some more background info on the design, DAC choice evaluation, DSP requirements etc:
https://paper.dropbox.com/doc/Phaser--AnD485kbUiDhEmmZ0cv8NUlZAg-yC4c4GTJ4hbA969VxIL9K

[image]

@gkasprow A way to talk to the flash via I2C would be really nice. But on Banker/Humpback this is really slow (a bitstream flashing operation takes some 5-10 minutes) and I think the I2C-SPI converter is the bottleneck (the way it limits and buffers each transaction). Maybe there is a simple way to speed it up or maybe we can figure out a different way to get this done over I2C.

@hartytp Didn't we say we'd use the DAC PLL to go from 125 MHz to 1 GHz DAC CLK? I didn't factor in the need to set up a HMC830 to clock the DAC.

hartytp, 2019-10-20:
> Also I'm not sure the Si5324 would be strictly needed for DRTIO. For the downlink it's not needed and for the uplink it may well not be required. And if there's a clock from MMCX/SMA then it's not needed for that either. I'd really try to minimize the clocking complexity for this.

Agreed, I don't want an Si5324.

hartytp, 2019-10-20:
> @hartytp Didn't we say we'd use the DAC PLL to go from 125 MHz to 1 GHz DAC CLK? I didn't factor in the need to set up a HMC830 to clock the DAC.

True, I'd forgotten about the internal DAC clock PLL. At one point we definitely discussed having 2 HMC830s, but it's very possible we agreed to scrap one later on and I forgot. Since we can steal the register set for the HMC830 from Sayma, I assume the additional work required to support a HMC830 for the DAC clock is basically just the usual work associated with hooking up an extra SPI bus?

In any case: (a) for the initial contract, the internal DAC PLL is definitely fine (b) I agree that we should try to minimise complexity/power consumption/cost as far as possible and should try to avoid using an HMC830 for the DAC clock (in any case, that's typically ~10-20dB less important than the mixer LO due to the frequency difference).

The things I would potentially be worried about with the internal DAC PLL would be noise and phase determinism (e.g. due to VCO auto cal). The latter may be solvable by using an active loop filter, fiddling with registers. I'll have a look at the data sheets/noise models and have a think. I'll open a separate issue on this.

jordens, 2019-10-20:
Iirc the DAC pll doesn't do any auto calibration. We'd set the vco range and be done. And I don't see how that would be affected by an active LF.
The DAC setup is pretty simple. We could freeze that and wouldn't have to implement some passthrough mechanism. The hmc830 does complicate things. it's pretty different.

jordens, 2019-10-20:
And the DAC34**H**84 would be sufficient. No S needed.

hartytp, 2019-10-20:
> Iirc the DAC pll doesn't do any auto calibration.

It may not; I haven't looked at the data sheet yet (just listing things that could be issues). AFAICT it's generally only the really high-performance ones that narrow-band step-tuned VCOs.

> And I don't see how that would be affected by an active LF.

Well, the question is why a step-tuned VCO would affect the output phase at all but empirically, it does! My understanding is that (a) the different bands on the same VCO can alter the leakage (b) the different bands change the DC bias voltage required to achieve a given output frequency, which (due to finite DC LF gain) requires a different DC offset at the phase detector. Putting an integrator between the PFD and VCO basically eliminates both effects.

The other thing I might worry about would be the PFD frequency for the DAC internal PLL (again, affects noise and drift). The 830 is nice as it can run directly at 125MHz. If there were any limitations that required us to run the DAC PFD at lower frequencies then I might be worried. but, again, this isn't something I've thought about yet, just a note to self for things to consider before finalising the design.

gkasprow, 2019-10-20:
Here is a very initial sketch. Just to estimate IO resources, power, clocking, etc.
[Phaser_SCH.PDF](https://github.com/sinara-hw/Phaser/files/3748520/Phaser_SCH.PDF)

jordens, 2019-10-21:
@hartytp Leakage and finite DC gain are the same thing in the loop filter. If there is no leakage, the DC gain is infinite and vice versa.
There is just one VCO and we need to do the coarse tuning (band select) ourselves in any case. The DAC provides a measurement of the LF voltage for that.
But sure, you can kill that effect with another integrator. I would suspect though that if PVT are sufficiently unstable to require an active LF to kill leakage-induced phase noise, you will typically also run out of the VCO fine tuning range because of PVT. And then the active LF is pointless.

jordens, 2019-10-21:
For IQ mixer, also check the [TRF3705](http://www.ti.com/lit/ds/symlink/trf3705.pdf). Better linearity, more output power, less noise, lower low-f corner, less bias, lower input power, more gain, but more power consumption, and lower high-f corner.

hartytp, 2019-10-24:
@gkasprow I looked over the design sketch. Looks good! I think the issues we've opened cover the main things. This time, let's try to finish the schematic review before doing work on the layout to avoid wasting time.

I'd like to send Stabilizer and Fastino to manufacture before beginning work on this in earnest. But maybe we can get to that some time next week? Depends on @jordens review timeline.

jordens, 2019-10-25:
I have already reviewed fastino and stabilizer as much as I'm involved. I'm not blocking those.
On phaser I can do some review next week. Between the issues and notes already filed and the design sketch there isn't much to be done.
Fpga wise an artix in that large pin compatible range from the a50t downwards will be fine (same as on sayma rtm).

hartytp, 2019-11-01:
@gkasprow I'm really keen to complete my initial design review soon. Could you have a go at producing a schematic only (no layout) RC please?

hartytp, 2019-11-05:
@gkasprow ping...what do you think the timeline for getting a schematic RC is? I'm itching to get this one off to manufacture and into my lab :)

gkasprow, 2019-11-06:
I plan to finish them until Friday.

hartytp, 2019-11-06:
🎆

closed 2019-11-06 by hartytp