sinara-hw/Phaser #2: external DAC clock PLL

hartytp, 2019-10-20:
e.g. if we're worried about phase indeterminism/noise from the DAC's internal PLL. Let's aim to avoid the external PLL unless we feel compelled to keep it...

gkasprow, 2019-10-20:
I'd use HMC830 and add the 0R bypass option. The DAC PLL can be bypassed with SW settings.

hartytp, 2019-10-21:
Let's go without the extra PLL for the first revision. The power budget looks scary with it. We'll test carefully when the boards arrive and can add an extra PLL for the next revision if we find the need.

I would like to have some way (even if it's only UFLs + solder jumpers) to inject an external 1GHz DAC clock for testing/debugging purposes.

jordens, 2019-10-21:
Agreed.

jordens, 2019-10-22:
The phase using the DAC PLL will be deterministic up to PVT coefficients.

closed 2019-10-24 by hartytp