sinara-hw/Phaser #3: DC bias for Mixer

gkasprow, 2019-10-20:
The ADL5375 requires DC bias. I assume we don't need DC-coupling between DAC and the mixer and adding RC biasing network would be fine.

hartytp, 2019-10-21:
Can't we just do something like this and use the DAC output for the bias?

[image]

gkasprow, 2019-10-21:
We probably can, providing that with this approach we will be within the common-mode range of the mixer.
I'll check it.

gkasprow, 2019-10-21:
The AD9122 has an output compliance range of -1 to +1V. With 20mA full-scale current this gives output range of 0...1V at 50OHm load.
We would need to use ADL5375-05 which accepts 0.5V input bias.
The DAC34H84 has the compliance range of -0.6 to +0.6V which means that the bias is 0.3V instead of 0.5V

gkasprow, 2019-10-21:
We will need a resistive level shifter, but this will attenuate the signal by roughly 50%.

jordens, 2019-10-21:
I think AC coupling is fine. Or select a different IQ mixer (TRF3705), that has compatible bias.

closed 2019-11-06 by gkasprow