sinara-hw/Phaser #4: DAC LPF

gkasprow, 2019-10-20:
I assume we need some LPF filter between DAC and the mixer. The LPF needs to be fully differential. We can go for this [one](https://www.minicircuits.com/WebStore/dashboard.html?model=DLFCG-540%2B), but there are not many options with different frequencies.

jordens, 2019-10-21:
Or just a discrete one, like for the AD9148. Maybe not even such an aggressive 5th order. Yes, needs to be differential.
[image]

hartytp, 2019-10-21:
@gkasprow the MCL filter isn't the right approach IMHO. It only comes in a small range of values, none of which are what we'd ideally want (with 1GHz clock and 540MHz, you don't get much filtering for some of the lower frequencies we'll use).

I suspect that a well-designed discrete filter with tight-tolerance components will be the way to go

hartytp, 2019-10-21:
@gkasprow how important is it to use a differential filter instead of, say, two single-ended filters?

gkasprow, 2019-10-21:
The differential filter has usually better matching between negative and positive terminals. Moreover, depending on the construction, it may attenuate mainly differential components while two filters will attenuate both diff and common-mode components. Some differential filters are two single-ended ones in one package. So there is no simple answer to the question.
I don't think it is that important in our case.

hartytp, 2019-10-21:
@gkasprow do you know of any other sources of any other manufacturers of SMT RF differential filters. I didn't see any at a look (but I might have missed one).

For the upconversion, I guess we'd really want a double differential filter for the I and Q channels to maintain amplitude and phase balance. So, it's not so clear to me that 4xsingle-ended is really worse than 2xdifferential.

Unless you're worried about this, I suggest we just go for standard MCL single-ended filters. I'd expect that the matching between filters will be better than the matching between discrete filters.

gkasprow, 2019-10-21:
I'm not sure if lTCC would have tighter tolerance than a classical filter composed of 1% LC components.
They don't even specify the tolerance in DS.

hartytp, 2019-10-21:
Ok. Are you happy designing a discrete differential LPF for this then? Since it needs to work to 1GHz, we'll have to be careful to simulate the parasitics correctly to get something that performs as expected.

jordens, 2019-10-22:
I would also go for discrete. Third or fifth order like in the data-sheets and eval boards.

gkasprow, 2019-10-23:
I will simply copy the eval board layout.

gkasprow, 2019-11-06:
I plan to use this filter
[image]

hartytp, 2019-11-06:
Cool!

Questions:
- the dac/modulator there aren’t the ones we’re using. Does that matter (different parasitics)
- will you use the same BOM lines as the eval board to keep the parasitic the same?
- how much will the layout affect the response?
- do you have a curve of the filter response? If not, can we do a quick sim with all component models
- are you using components with very tight tolerances (eg the i and q channels need really good matching to preserver the mixers excellent side and suppression)

gkasprow, 2019-11-06:
I'm using 2% inductors and capacitors in the 0603 packages to make modifications easier. I will remove the lower layers of PCB to limit the parasitics. We are talking about 2 components 68nH and 2.7pF. In the case of the capacitor, the influence of its parasitic capacitance is critical and I'm not sure if we can precisely simulate such a circuit. EVKIT does not have any filter.
The DACs have current outputs that are loaded with 50R. It's hard to say if its parasitics matter that much.

gkasprow, 2019-11-06:
Anyway, I will estimate pads capacitances and do the simulation.

gkasprow, 2019-11-07:
The very first approach
[image]
and the charasteristics:
[image]

gkasprow, 2019-11-07:
This schematic includes all parasitic components
[image]
[image]

jordens, 2019-11-07:
ok. @hartytp I think this is probably worse than 0.1 dB flatness over 20 MHz already at 300 MHz.

hartytp, 2019-11-07:
> ok. @hartytp I think this is probably worse than 0.1 dB flatness over 20 MHz already at 300 MHz.

Indeed. 0.1dB is better than we need (just 1dB is a bit on the high side, but nought-dot-a-few-dB is ok).

hartytp, 2019-11-07:
Thanks greg, that looks pretty good. What are the inductor part numbers?

closed 2019-11-07 by hartytp

gkasprow, 2019-11-07:
Coilcraft 0603HP-68NXGLU