sinara-hw/Phaser #5: Mixer bypass

gkasprow, 2019-10-20:
do we need a switchable mixer bypass?

hartytp, 2019-10-21:
No. Far too much effort to put in RF switches. Let's just make it a population option.

jordens, 2019-10-21:
Yes. Just 0R/capacitor jumpers.
It might make sense to maintain the baseband (no mixer) configuration as a different variant since the difference in population (PLL, fanouts, mixers, filters, biasing, new power rails etc) and cost is probably significant.

hartytp, 2019-10-21:
> It might make sense to maintain the baseband (no mixer) configuration as a different variant since the difference in population (PLL, fanouts, mixers, filters, biasing, new power rails etc) and cost is probably significant.

Yes, definitely.

Ideally, I'd also like a FP indicator (text/LED/whatever) to show which population variant one is using.

dtcallcock, 2019-10-28:
Bit late to the Phaser party, but anyway I'm very interested in using it for a new project.  However, I will need to be able to run down as low as 1MHz on the mixer bypass variant.  Is there a low frequency cutoff in the spec anywhere (I couldn't find one)?  If not, can we make 1MHz the spec?

The only issue I see right now is that the 10nF DC-blocks are a bit weedy.  Perhaps they could be 100nF in this variant?

jordens, 2019-10-28:
The balun (TLT vs not) will be the bigger hurdle for low f.

jordens, 2019-10-28:
Well, depends on how we do it. Whether the balun would bypass the mixer and whether the mixer is  unbalanced.

dtcallcock, 2019-10-28:
Yes, it seems that if this is a population variant there is quite a bit of flexibility.  I think I need to see a sketch of where we are right now though as I'm having a hard time picturing it from the comment threads.

hartytp, 2019-10-28:
@dtcallcock that should be possible. The magnetics are the things that are most likely to cause a problem. But I suspect we can find a way of making this work.

gkasprow, 2019-10-28:
@dtcallcock  I assume you want a circuit like in Urukul. We can make the balun assembly option so you could use one that works from 400kHz like [ADT1-1WT+](https://ww2.minicircuits.com/pdfs/ADT1-1WT.pdf). Anyway, this one works up to 800MHz so it should satisfy other applications as well.

hartytp, 2019-10-28:
Yes maybe that for the bb variant and a broadband tlt for the upconverter?

dtcallcock, 2019-10-28:
That works.

jordens, 2019-10-29:
We may get around having to use a transformer for.the upconverter variant.

hartytp, 2019-10-29:
How?

jordens, 2019-10-29:
There's no need for a transformer between mixer and upconv if the bias and matching work out (see data-sheets). And after the mixer it's already single ended.

jordens, 2019-10-29:
But for this application I am pretty sure that mixing down after phaser is actually the easiest approach. Then you can go all the way to DC.

gkasprow, 2019-10-29:
@jordens do you mean to use mixer as a diff to single ended converter with LO switched off ?

jordens, 2019-10-29:
I believe we'd use

* Baseband variant: dac, diff discrete lowpass, balun, digital attenuator, switch. The the balun could be the standard ADT 1-1 or maybe a 4:1 for a bit more power if that works with the matching
* Upconverter variant: dac, diff discrete lowpass, mixer, digital attenuator, switch.

jordens, 2019-10-29:
But yes. If you want to go all the way to DC, I'd use the baseband variant, and slap an external downconverter (125 MHz LO e.g.) after it.

hartytp, 2019-10-29:
@jordens yes, thank you, I'd forgotten that trf372017 has differential I/Q inputs, but single-ended RF output. So, we shouldn't need any balun for the upconversion path, which simplifies things. FWIW since the upconverter has internal DACs for bias control, I'd just AC-couple the DAC outputs in all cases.

[image]

hartytp, 2019-10-29:
(as you say though, we need to check that matching doesn't require a balun, so need to check data sheets/eval boards before making a decision)

hartytp, 2019-10-29:
> I believe we'd use

Yes, completely agree. We can potentially scrap the switch as well to get some more output power/reduce complexity, but I'm fine keeping it...

NB if we use the DAC amplitude rather than the switch to turn the output on/off then we'll need to rethink the noise eater control logic...

hartytp, 2019-10-29:
> But yes. If you want to go all the way to DC, I'd use the baseband variant, and slap an external downconverter (e.g. 125 MHz LO, Phaser into RF and use IF output) after it. It's not the ultimate solution in terms of low frequency noise but then again for DC applications other issues (like DAC tempco 15-30 ppm/K) might be problematic anyway.

I currently don't see any use-case for DC on this design. Without knowing the use-case, it's hard to comment on suitability, but I generally suspect this is not going to be the right design for dc-sensitive applications. e.g. may be better to use a bias-T to combine Fastino and Phaser outputs or something like that. Anyway, this is outside the scope of this design and a little off-topic.

jordens, 2019-10-29:
Yes. Let's kill the switch as well. If 40ns resolution is fine, then a virtual "TTL" channel to blank the output would be easy.

hartytp, 2019-10-29:
> Yes. Let's kill the switch as well. If 40ns resolution is fine, then a virtual "TTL" channel to blank the output would be easy.

I only imagine using phaser in cases where we need pulse-shaping, so the switch wouldn't be used anywhere in our setup (well, maybe to kill LO leakage during long off periods, but that's a bit niche).

Not sure what the right way of blanking the noise eater is. Maybe we just need to manually enable/disable it at the start/end of each pulse, but that's a separate conversation.

jordens, 2019-10-29:
Or we can just use the TXENA pin which presumably has either data clock resolution (2ns) or dac clock resolution (1ns). This is would be good for noise eating as well.
Unfortunately it affects both channels.

hartytp, 2019-10-29:
> Or we can just use the TXENA pin which presumably has either data clock resolution (2ns) or dac clock resolution (1ns). This is would be good for noise eating as well.
Unfortunately it affects both channels.

Well, worth hooking up the FPGA at least for future use.

jordens, 2019-10-30:
For silencing the output (LO leakage) there's also PWD_TX_DIV in the upconverter accessible through SPI and AFAICT without sideeffects.

gkasprow, 2019-11-06:
I got the mixer devkit. If som measurements are needed, I can make them.

gkasprow, 2019-11-06:
I left the RF switches.

hartytp, 2019-11-07:
Where are they? I don’t think we can have a switch on the I as it will destroy the I/Q matching. Let’s just use solder jumpers

jordens, 2019-11-07:
I think he meant the RF switch after the mixer.

hartytp, 2019-11-07:
@gkasprow either way, let's drop it. Unnecessary insertion loss and complexity and we can quickly turn the DAC off in software so it's not helpful to have a separate switch

gkasprow, 2019-11-07:
I meant this circuit
[image]
OK, let's get rid of them.

closed 2019-11-07 by gkasprow