sinara-hw/Phaser #7: Config FLASH update

gkasprow, 2019-10-20:
> A way to talk to the flash via I2C would be really nice. But on Banker/Humpback this is really slow (a bitstream flashing operation takes some 5-10 minutes) and I think the I2C-SPI converter is the bottleneck (the way it limits and buffers each transaction). Maybe there is a simple way to speed it up or maybe we can figure out a different way to get this done over I2C.

What about making SPI FLASH available on the LVDS lines?
One can achieve it in a few ways:
- add LVDS mux that switches 4 EEM lanes to the FLASH. It is controlled by I2C. Simplest and robust. LVDS muxes like [this ](https://www.maximintegrated.com/en/products/analog/analog-switches-multiplexers/MAX14979E.html) They do not cause additional lane delays.
- make the FLASH inerface inside of the FPGA as it is done for JTAG FLASH programmer. Needs some gateware development and is not fail-safe. One cannot recover in case of broken flash content
- use existing I2C to SPI converter + add  tihgtly coupled, slow LVDS receivers that work only for FLASH writing. I2C to SPI works for status/readout. In this way we avoid switching the LVDS bus but it may cause issues and complicates the things since require both I2C and SPI access. 
- make a CPLD  I2C to SPI bridge that utilises fast I2C mode.

jordens, 2019-10-21:
(1) is pretty complex on the coding side and would need quite some development on Kasli and then it's as easy to do (2) instead which doesn't need additional hardware. (3) also requires additional tooling, gateware and scatters some complexity across different interfaces. My impression is that (4) would also require quite some development work for not that much gain and reusability.

I would keep the existing I2C-SPI solution and go for (2) at some later point.
There are good ways to make it (almost completely) fail-safe with multiple bitstreams in flash and warm booting.
And we already have two decent, robust solutions (JTAG cable and I2C-SPI). If we can't implement (2) then they are good enough for me. Waiting 10 minutes for a firmware update is not that dramatic at the user end. Developing with JTAG connected is a must in any case since flashing is still about 2 orders of magnitude slower than loading a bitstream via JTAG. No point in wiring up another FTDI solution here.

gkasprow, 2019-11-06:
done.
[image]

jordens, 2019-11-07:
Is the FPGA's CFG_CCLK high-Z during INIT_B low? I'm not sure since it's dedicated.

jordens, 2019-11-07:
I believe the answer to the above is yes.

But wouldn't it be useful to connect PROGRAM_B to a I2C-SPI GPIO (and either derive FLASH_UP_EN from INIT_B or to drop the FPGA_DONE input)? INIT_B alone doesn't reload the configuration.

gkasprow, 2019-11-07:
Right. Do we need INIT_B at all?

gkasprow, 2019-11-07:
OK, to hold FPGA configuration one needs to hold init_B low. TO restart configuration one needs to toggle PROGRAM_B. So, what if I connect them both with two Schottky diodes to GPIO?
In this way I will pull them low together.
[image]
[image]

jordens, 2019-11-07:
Schottky sounds good. Just make sure that they can pull low enough against the pullups.

gkasprow, 2019-11-07:
DONE line has 300R pullup so I will add two MOSFETs.
[image]

closed 2019-11-07 by gkasprow