sinara-hw/Phaser #9: LO filter

gkasprow, 2019-10-21:
The HMC830 has a digital output which means a lot of harmonics.
Shall we add an LPF?
[image]
DS specifies even recommended RC values
[image]

hartytp, 2019-10-21:
> The HMC830 has a digital output which means a lot of harmonics.

Isn't that what one want's for a mixer LO?

hartytp, 2019-10-21:
Okay, good spot, that's new to me...

[image]

jordens, 2019-10-21:
Needs to be sinusoidal because of the 90 deg hybrid. The mixer comes after that but still needs enough power.

hartytp, 2019-10-21:
Still, that's quite annoying, as it will require a different population option for each LO frequency, which makes the board far less versatile.

hartytp, 2019-10-21:
>Needs to be sinusoidal because of the 90 deg hybrid. The mixer comes after that but still needs enough power.

Yes, that's what I took from the data sheet section. I actually have never read up on how broad-band hybrids are produced and don't really understand it.

gkasprow, 2019-10-21:
We can use external VCO that gives sine output.  It won't be that wideband but there are VCOs that are available in many frequency ranges.

gkasprow, 2019-10-21:
for example IC from [this family](https://www.analog.com/en/products/rf-microwave/vcos-plos/vcos-with-single-output.html) + PLL chip

hartytp, 2019-10-21:
> There is Mirny.

Still requires a custom harminic suppression filter that's specific to the LO frequency. And still has the awkwardness of converting the differential signal to single-ended and back again. Much more expensive as well (a 1k board when a $10 IC will do). 

Maybe the thing to do here is to put pads for 2xMCL single-ended SMT LPFs (since differential ones are hard to come by). Pick a sensible default option, e.g. a 5GHz cut-off (so, good down to ~1.7GHz LO), and let users specify other filters as required.

jordens, 2019-10-21:
Agreed.

hartytp, 2019-10-21:
> We can use external VCO that gives sine output. It won't be that wideband but there are VCOs that are available in many frequency ranges.

True wideband VCOs tend to have very bad noise. We can try to find a family with good coverage, but IME, it's hard to find a family of low-noise VCOs that have good spectral coverage (which is why the step-tuned narrow-band VCOs are such magic).

I think I'd prefer (a) users pick filter population options for their choice of LO (we're only trying to suppress third harmonics, so we can cover the whole ~1-6GHz range of the mixer in a few filter choices) (b) use an external LO and let users select a coaxial (SMA) filter.

hartytp, 2019-10-21:
The other option is to go for something like a [HMC1197] with integrated LO, and then one doesn't have to worry about this. Also has the added perk of making the two channels fully independent. We'd want to look at cost/power consumption first, but worth considering.

One downside is that I'm not aware of any modulator with integrated LO that goes above 4GHz, so we might end up loosing a bit at the top end of the spectrum -- unless that's already limited by something else like the magnetics...

hartytp, 2019-10-21:
From a quick skim over the datasheets (may contain mistakes)...

2 channels of HMC1197 is 3.5W (which is appreciable!).

HM830 is about 1W depending on settings. ADL5375 is something like 1W/channel. So the power consumption for the two approaches doesn't look radically different. Cost looks comparable as well. Also removes the need to distribute a LO around the PCB -- just a single quad LVDS buffer for the 125MHz, which may simplify layout

hartytp, 2019-10-21:
TRF3705 is also only 4GHz max. So actually not different to the HMC1197.

There are also other parts like TRF372017 which also have in-built VCOs.

Needs a careful look before making a decision, but I'm starting to think that having two modulators with in-built VCOs may be the easiest way to go.

jordens, 2019-10-21:
A true wideband VCO (a YIG) can be very good in terms of noise.

gkasprow, 2019-11-06:
nothing to do here, closing

closed 2019-11-06 by gkasprow