sinara-hw/Phaser #10: choice of upconverter

hartytp, 2019-10-24:
I'd like to aim to use upconverters with inbuilt LO PLLs. The thinking here is that it's highly desirable to have a LO PLL on the EEM, rather than relying on an external LO. Generating the LO with a PLL outside the modulator is mildly tricky because the quadrature hybrids in the upconverter don't like harmonics, which most PLLs produce. Filtering the harmonics is a pain. Easier to let the modulator designer take care of that. Having the PLL inside the LO also makes the two channels fully independent, which may or may not be useful for a given application, and means we only have one clock frequency to distribute (the reference)/saves us a front panel connector.

The potential downsides of using a modulator with inbuild PLL/VCO are:
- power consumption -- but, doing the maths, I don't think having two PLLs/VCOs uses much more power than having an external PLL (at least, for a high-performance one like an HMC830)
- less choice of modulator.

@gkasprow do you know any manufacturers of upconverters with PLLs/VCOs apart from TI and ADI? I'll post a table here with our options soon.

hartytp, 2019-10-24:
PN | BW (GHz) | power (W) | supplier |
---| ---| --- | --- |
[trf372017](http://www.ti.com/lit/ds/symlink/trf372017.pdf) | 0.3-4.8 | ? | TI
[trf3722](http://www.ti.com/lit/ds/symlink/trf3722.pdf) | 0.4 - 4.2 | ? | TI

hartytp, 2019-10-24:
*TRF372017*
- internal bias
- VCO output, route to MMCM/UFL for debugging (can be DNP by default)
- 200mA from 3V3, 117mA from 5V = 1.245W
- f_pfd max: 100MHz (so we will run at 62.5MHz)

VCO at 2GHz, noise normalised to 1GHz

f offset | HMC830 noise (-dBc/Hz) | trf372017 noise(-dBc/Hz)
--- | --- | ---
10k | - | 92
100k | 122 | -
1M | 147 | 139
10M | 168 | 157
50M | 177 | 160

So the VCO is something like 10dB worse than the HMC830, which is fine.

- PLL figure of merit is the normalized phas enoise floor of the PFD in dBc/Hz. i.e. measure in-band phase noise (closed loop) and subtract `20log(N) + 10log(f_pfd)`
- HMC830 is (integer-N) is -230dBc/Hz. So, at 2GHz, f_pfd=62.5MHz that gives
  - N = 32
  - `PN = -230dBc/Hz + 20*log(32) + 10*log(62.5) = -122dBc/Hz`
- For TRF372017 (figure 21)
  - `f_pfd` = 1.6MHz, `f_lo_out = 1GHz` so `N = 625`
  - `FOM = -103 - 20*log(625) - 10*log(1.6e6) = -221`
- So, again, the TRF372017 PFD is about 10dB worse (I can't find a flicker spec for this PN)
- Bearing in mind that the 830 is about the best on the market for PLL with integrated VCO, that's good!

hartytp, 2019-10-24:
[image]

Worse case variations are of order 0.2dB/60K = 0.07%/K. From the plots, it's generally quite a bit better than that (a lot at certain frequency/temp combinations)

hartytp, 2019-10-24:
[image]

hartytp, 2019-10-24:
[image]

hartytp, 2019-10-24:
trf3722
- seems broadly comparable.
- VCO bb noise is a bit better, close in noise a bit worse. PFD FOM is about identical (-221dBc/Hz)

[image]

hartytp, 2019-10-24:
HMC1197
- 320mA*5V + 48mA*3.3V = 1.75W
- PFD FOM is -230dBc (integer N)
- VCO is pretty close to HMC830

- NB I assume the TI FOMs were integer N, but wasn't specified in data sheet, could have been fractional-N (but I doubt it)...

hartytp, 2019-10-24:
My feeling is that we should go for the trf372017. Covers a good frequency range. Not too power hungry. Decent noise. Well specified, nice internal biasing.

@gkasprow @jordens any objections/other suggestions, or shall we freeze that choice.

jordens, 2019-10-25:
The advantage of a shared pll would be differential phase if we don't use integer n.
Other than that the trf372017 is good.

hartytp, 2019-10-25:
Okay, if it’s not hard to route, let’s hook ch0 LO our to ch1 LO in. Then ch1 LO our to mmcx. That provides an option for this

gkasprow, 2019-10-25:
@hartytp did you look at Maxim ? They have quite a few up/downconverts but I'm not sure if they work above 3GHz.
Other vendors
Quorvo
NXP
Macom
IDT

Interesting ICs
https://pl.mouser.com/datasheet/2/412/ffc5071_5072_data_sheet-1506879.pdf
https://pl.mouser.com/datasheet/2/412/f2054_data_sheet-1110962.pdf

hartytp, 2019-10-25:
Thanks @gkasprow! I'll have a look at a few of those before Monday. Although, I'm already quite happy with the trf372017, so I'll need to find a good reason to switch.

hartytp, 2019-10-25:
From Quorvo, the possible choices are https://www.qorvo.com/products/frequency-converters/integrated-synthesizers-with-mixers#ta0050;;aa0112:2,aa0112:2;/

They are generally quite nice (I've never seen a chip like that with an active LF before). 

However, the thing I don't like about them is the mixers. They're not IQ mixers, which is inconvenient as we'd have to externally filter the unwanted sideband. And, the input/output coupling requirements for the mixers look more annoying than the TI chips.

So, nice chips, but I don't see a significant win over the TI offerings.

hartytp, 2019-10-25:
I don't see any Macom upconverters with integrated LOs, so let's rule them out.

hartytp, 2019-10-25:
Also, AFAICT, the Quorvo chips have to be bought directly from the manufacturer which, I imagine, is annoying for the assembly house https://octopart.com/search?q=ffc5071&sort=median_price_1000&sort-dir=asc

Do you know of any reason we'd prefer those to the more readily available TI offering?

hartytp, 2019-10-25:
I don't see any directly comparable NXP products.

hartytp, 2019-10-25:
I don't see any IDT parts in a comparable frequency range, so that rules them out too (unless I'm missing something -- the way these companies sort the parts always confuses me!).

So, thanks a lot for the links @gkasprow but AFAICT the trf372017 is still the best option.

Unless anyone has anything else to add, let's fix that design choice.

That's the component identification / circuit topology phase complete! Next stage is schematic review. Whoo!

hartytp, 2019-10-25:
The RFFC5071A VCO/PLL noise seems worse than the TI part.

[image]

FOM: Phase noise is around -100dBc/Hz with 26MHz REF and 1.5GHz output. N = `FOM = -100dBc/Hz - 20*log10(58) - 10*log10(26e6) = -210dBc/Hz`

hartytp, 2019-10-25:
Power consumption is only 330mW or so, which is a lot lower than the other modulators. But I think that's mainly because it's not a proper IQ modulator.

Let's go with the TI chip

gkasprow, 2019-10-25:
That's fine with me. You asked about another options so you got them :)

hartytp, 2019-10-25:
Indeed, thanks for the links - always good to see more options

gkasprow, 2019-10-28:
@hartytp one remark. Correct me if I'm wrong, but we decided to use PLL + mixer combo because external PLL like HMC830 delivers square waves and requires filters. But this combo chip does exactly the same. You will still get plenty of harmonics without the output filters. I didn't notice any filters on the block shematic of the ICs. Such filters would need to be bulky and tunable in wide range.

hartytp, 2019-10-28:
@gkasprow thats a good question and one I was thinking about too. Short answer is that the ic designers take responsibility for this issue and how to solve it. The ic performance is well specified so clearly they have some solution. Not sure how they do it!

gkasprow, 2019-10-28:
Are you sure they specify the IC for operation with such wide bandwidth? These chips are usually used for relatively narrowband circuits. I'd buy devkits of these ICs give it a try. This looks too good to be true:)

hartytp, 2019-10-28:
The spec is pretty clear afaict. Have a look and see what I’m missing. I can buy a dev kit and do some tests, but I’m not sure what I’d be looking for. Conversion gain variations? Sb suppression?

jordens, 2019-10-28:
I can imagine that with an integrated vco the quadrature generation can be done in a different way than with the 90 deg hybrid.

gkasprow, 2019-10-28:
@hartytp correct me if I'm wrong.  You want to shift the entire IQ bandwidth of -300 ..+300MHz up to let's say 3.2GHz +/- 300MHz. With such upconverter IC you will get also products at 9.6GHz +/- 300MHz which can be easily filtered away. So no issue here.
But if somebody wants to convert it to let's say 1GHz +/-300MHz, the products will be also at 3GHz and 5GHz.  THat's the only thing I'm worried about. I have an impression the DS does treat it as an obvious thing. ADI has also a [note ](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=10&ved=2ahUKEwixk4P587_lAhVNR5oKHbIIACYQFjAJegQIABAC&url=ftp%3A%2F%2Fftp.analog.com%2Fpub%2Fcftl%2FADI%2520Classics%2FBasic%2520Linear%2520Design%2520(Linear%2520Circuit%2520Design%2520Handbook)%2C%25202007%2FChapter_4_RF_IF_Circuits.pdf&usg=AOvVaw3Trpz7RxZD3LtwJowPQGCc)about such switching mixers.

hartytp, 2019-10-28:
@gkasprow right. There are two separate issues here:
- for a standard IQ upconverter, harmonics on the LO mess up the LO quadrature hybrid and can seriously impact performance. For this reason, the typical recommendation is the 3rd harmonic to be something like -30dBc, which is tough when using an external wideband step-tuned PLL
- beyond that, there will doubtless also be low-level output spurs at harmonics of the LO. That's the case even with a standard passive mixer. Again IIRC, it's well specified on the IC data sheet. This may or may not be an issue depending on the use-case. Since we have no pre-amp and won't operate any part near saturation, it's easy for the user to filter these off later on (e.g. with an MCL SMA LPF). This is something I'd think of as a standard RF issue that people need to be aware of.

gkasprow, 2019-10-28:
That's true.  Without the wideband hybrid, we get only 3rd and 5th harmonics which are easy to remove.
Anyway, I will buy the devkit to have the GUI and interface to quickly characterise the Phaser board once it's ready, without waiting for ARTIQ support.

hartytp, 2019-10-28:
Exactly.

Sounds like a good plan.

Now all critical components are selected, what do you think the turn around time for this project is likely to be?

gkasprow, 2019-10-28:
well, it depends on priorities,  Phaser or Pounder can wait ?

hartytp, 2019-10-28:
From my end, Phaser is by far the top priority. Others may feel differently ;)

hartytp, 2019-10-28:
I have time to do design reviews right now, so I can promise to turn things around quickly :)

gkasprow, 2019-10-28:
I have other high priority design to finish. I plan to do it by Wednesday. Then we have holidays so will be away. I will work on it next week. Anyway, the end of next week (Nov 10) seems to be feasible, I will do my best to publish the files for review earlier.

hartytp, 2019-10-28:
That would be awesome! Although if time is tight, finishing booster remains my number 1 priority over all

dtcallcock, 2019-10-28:
> Phaser or Pounder can wait ?

I'm happy for Pounder to move down the queue for Phaser.

closed 2019-11-06 by gkasprow