sinara-hw/Phaser #11: clocking notes

hartytp, 2019-10-24:
Copy from Urukul:
- 125MHz from FP SMA, MMCX or on-board XO
- high-quality LVDS fanout to DAC/modulators
- MMCX + solder jumpers to allow using an external LO (for debugging)
- route modulator LO output to MMCX (DNP by default) for debugging
- PLL locked indicators to RED LEDs on FP

jordens, 2019-10-25:
I'd be happy to drop the xo. And the associated mux in this case. It would require special consideration if the DAC clock is not locked to the link clock.

jordens, 2019-10-25:
Note that I haven't checked phase determinism with an external DAC clock bypassing the pll.
Also if the dac/lo pll don't have hardware pll lock indicator pins, then routing those signals to fpleds will be tricky.

hartytp, 2019-10-25:
Ack re ext dac clock. Afaict it’s not needed in any case (if the fpga transceiver pll is good enough for our ref clk l, the dac pll should be fine!)

hartytp, 2019-10-25:
Still nice to have a way of injecting an ext dac clock during debugging though even if it leads to phase indeterminism

gkasprow, 2019-11-06:
I kept ext clock, removed oscillator. TRF chip, GTP and DAC PLL are clocked from same clock mux outputs

gkasprow, 2019-11-06:
lock outptus are routed to FPGA, as well as LEDs

closed 2019-11-06 by gkasprow

hartytp, 2019-11-06:
Perfect

gkasprow, 2019-11-08:
@hartytp the schematic is ready for review
https://github.com/sinara-hw/Phaser/releases/tag/v1.0rc1

hartytp, 2019-11-09:
Amazing! Thanks

Will review tomorrow