sinara-hw/Phaser #13: design notes

hartytp, 2019-10-25:
This is largely a repeat of @jordens notes, but here are my thoughts on the motivation for the design choices here:
- There are two main limitations this design has to deal with: thermal/power and bandwidth
- Bandwidth:
  - Assume we have a Kasli-EEM link with 6 data lanes, 125MHz DDR. That's 12 bits per RTIO clock cycle (1.5Gbps). In practice we loose a little due to framing, CRC, etc
  - For 16-bit DAC words, two channels, we need 64 bits per IQ sample. So, 6 RTIO cycles per sample, which is +-20.8MHz nyquist bandwidth. Assuming 80% usable BW after filtering, that gives +-16.7MHz analog bandwidth
  - That's a bit marginal for us. We can increase it in a few ways:
    - go to 14 bit DAC data, so we're now at 5 RTIO cycles per IQ sample, 25MSPs IQ samples or +-12.5MHz nyquist bandwidth or approx +-10MHz of usable analog bandwidth. That's just okay for us, so this is what we chose. 
    - we could drop the clock line on the EEM to allow us to go to 7 data bits (use the MMCX clock for the data). This would make the data link more complex, but would allow us to get back to 16 bits at the same sample rate. 
    - we could use two EEM connectors. This gives more BW, but means fewer channels per Kasli and makes the link more complex (more skews to compensate for)
    - we could try to increase the link clock rate. Working at non-multiples of the RTIO clock is messy, not sure if 250MHz will work due to SI (reflections, cross-talk, attenuation in the ribbon cable) would need testing
  - decision was to go for 20MHz BW with 14 bit samples
  - the samples written to the DAC are still 16-bit. We will do intensity stabilization/noise eating on the EEM FPGA and convert from 14-bit to 16-bit in the process. This allows us to do the intensity stabilisation without loosing bits, which is nice
  - the plan is to use an up converter (NCO + digital mixer) on the EEM to mix this bandwidth up to a chosen frequency. The NCO can be tuned phase coherently in ~1us.
  - Kasli generates the IQ data. Plan is to have 4 tones per channel in the initial implementation (all share the same NCO so must lie in the same 20MHz bandwidth centred on the carrier)
  - the DAC can take samples at up to 750MSPS (1.5GHz clock, 2x interpolation). We're adding RAM to the EEM to allow it to operate as an AWG as well (gateware not funded). e.g. write samples and trigger playback via the EEM connectors
- Thermal:
  - Urukul is around 7W (max from data sheet values is more like 10W, but it doesn't consume that in practice). We don't want to go much higher than this to avoid thermal management issues
  -  That rules out things like JESD DACs and large FPGAs. need to limit the computational heavy lifting carried out on the FPGA. Preferable to use a DAC with inbuilt NCO as this tend to be much more power efficient

Since the DAC has IQ outputs, the plan is to have two modes of operation, selectable by population operations: baseband (DC to ~400MHz) and up conversion (via an analog IQ mixer) giving ~400MHz to ~4GHz

jordens, 2019-10-25:
No point in buying the DAC34SH84. The Artix 7 pins can only do 1.25 Gb/s at best.

hartytp, 2019-10-26:
Yes, thanks. I was thinking of a different dac pn that only had one parallel port, so 1 pair of IQ samples per 2 dac clock cycles. In that case we could have run at 1.5GHz for 750MSPs

hartytp, 2019-10-26:
Anyway nice to keep 1GSPS AWG as an option

jordens, 2019-10-27:
We would also do 1 sample per 2 DAC clock cycles. There are three different pns. The narrow one, the wide one and the fast+wide one. We want the wide one.

closed 2019-11-06 by hartytp