sinara-hw/Phaser #14: mixer sideband suppression, and carrier nulling

jordens, 2019-10-30:
The strategy to do this should be thought through now. This will be frequency/power/temperature dependent and will need to be done by the user.

Anything simpler than downconverting (LO amplifier + mixer + lowpass) one channel using the other channel and using the on-board ADC? Write software to do this, maybe map that and add to coredevice driver.
Or do we want to on-board this so it can be done without the required 12 SMA disconnect/reconnect cycles?

hartytp, 2019-10-30:
Thanks @jordens. This is something I've considered but not written up yet.

The uncompensated sideband and carrier suppression is

[image]

[image]

So, the first question is whether the applications we have in mind for this design need anything better than those unadjusted numbers. 

I'd be interested in input from other users -- so far the input I've had has been "we don't need upconversion -- but speaking for our work, the answer is that the unadjusted numbers are fine. These are two very specific spurs, whose location can be controlled to a certain extent by appropriate choice of LO, and those levels are fine for us.

As you note, the calibration is nasty as it's non-linear, frequency dependent, etc and the benefit for our work is too small to bother

hartytp, 2019-10-30:
I'd also be wary of adding extra hardware to phaser to cover this unless someone presents a strong use-case. I feel we're a bit marginal on power consumption already and I'm generally wary of complexity/cost for these designs.

hartytp, 2019-10-30:
The other thing I'd add is that it's likely that the PVT variations are quite a bit smaller than the dependence on the LO frequency. As such, for a given LO frequency, it's likely one can calibrate one channel of one board and then determine appropriate Vcm etc to use. That would be a per-use (per LO frequency) rather than per channel calibration, so the number of SMA connections/reconnection is less important.

gkasprow, 2019-10-30:
The power of the feedback circuit can be disabled entirely when not needed.

jordens, 2019-10-30:
I could imagine that when jumping around in a Forrest of hyperfine/Zeeman transitions the unwanted sideband and carrier might be problematic. It certainly depends on the application though.
With the external or internal setup your be able to map out the full frequency/Power space and tune balance and offset. That's how e.g. the gnuradio boards do it.
The internal one would need a coupler, (simple) mixer, and lowpass. We could use the lo outputs and cross the channels. Either with Rf switches and a single downconversion chain or duplicating the downconversion circuit.

jordens, 2019-10-30:
Still, to do the mapping with the external circuit it would be comparable to the connector juggling you have to do when calibrating a vna.

hartytp, 2019-10-30:
> I could imagine that when jumping around in a Forrest of hyperfine/Zeeman transitions the unwanted sideband and carrier might be problematic. It certainly depends on the application though.

Sure, I can imagine situations where it would be an issue. For the experiments I have in mind, I'm confident it won't, and I'm wary of designing for a use case I don't fully understand.

> With the external or internal setup your be able to map out the full frequency/Power space and tune balance and offset. That's how e.g. the gnuradio boards do it.
> The internal one would need a [...]

Ok. If you have a design in mind that's low cost/power/complexity and you think will expand the user-base for the design then I don't object to adding it (although, I don't think I'll be able to provide much input for the design or any additional funding).

jordens, 2019-10-31:
If there is no interest in doing this then we can close it. Investing in it is too risky for me.

hartytp, 2019-10-31:
Ok. As I said I can see the value for some use cases, but it wouldn’t be particularly helpful for us and I’m reluctant to design for something without a concrete use case.

We can always add this in a future design revision if it becomes important for someone

closed 2019-10-31 by hartytp