sinara-hw/Phaser #15: Power sequencing schematic annotation

dnadlinger, 2019-11-09:
I was just flipping through the v1.0rc1 schematics out of curiosity, and noticed that the power sequence annotation in PWR_Supply-5.5V.SchDoc only lists 4 steps, but the analogue voltage rail regulators aren't enabled from (P5V5 & P1V0) like P1V5/P3V6 are, but from P1V5.

Are they intended to come up after P1V5 (in which case the annotation could use clarification), or is this just due to some signal level quirk I didn't notice (in which case just ignore this)?

gkasprow, 2019-11-09:
the sequence should be:
- ADC 5VA
- reference
- AFE

dnadlinger, 2019-11-09:
Ah, I was just wondering why EN on IC19, IC21, and IC25 is tied to P1V5 rather than together with IC14/IC32, as the rails all those produce are listed in the same sequencing step.

dnadlinger, 2019-11-09:
Ah, I was just wondering why EN on IC19, IC21, and IC25 is tied to P1V5 rather than together with the EN of IC14/IC32, as the rails that all those produce are listed in the same sequencing step.

gkasprow, 2019-11-10:
OK, this is to make sure that 3V6 and 1V5 wake up first, deliver supply to the DAC before FPGA wakes up. This is strictly required by the DAC.
Since VCCINT wakes up first, we can enable the other rails in any order. That's why we need 1.5V to generate 1.2V required by DAC.
I can always add another 1.5V converter but this should work.

hartytp, 2019-11-11:
I find this annotation a bit confusing `P1V5, P1V2A, P3V6, P3V3A, ADC, P5V0A - DAC` Do you mean something like `P1V5, P1V2A, P3V6, P3V3A, P5V0A (ADC, DAC)`

hartytp, 2019-11-11:
NB we're applying a reference to the ADC OpAmps before they get their powers. This seems slightly non-optimal, but I guess it's okay since there is a decent sized resistor between the ref and the OpAmp so the current is limited.

hartytp, 2019-11-11:
Would definitely be better to power the AFE supplies *before* the reference, but still after the 5V0.

hartytp, 2019-11-11:
This could even just be done with an RC delay on the ref enable, I guess (with a diode to stop it staying enabled after the AFE is turned off during shut-down)

hartytp, 2019-11-11:
NB the ADC max input voltage is relative to the 5V0A rail, not the reference...

[image]

gkasprow, 2019-11-14:
I added 15k / 1uF delay but we don't really need it.

hartytp, 2019-11-18:
This is the RC delay that was added to keep the reference off while the AFE OpAmps start up.

[image]

hartytp, 2019-11-18:
Actually @gkasprow please can we remove these components? Thinking again, this is a really bad idea and we should have the ref come on before the +-12V rails. Otherwise the OpAmps can drive a negative voltage into the ADC which could damage it.

closed 2019-11-29 by gkasprow