sinara-hw/Phaser #18: FPGA selection/powering/banks

jordens, 2019-11-09:
* FPGA package is good
* maybe don't power bank 13 if we don't need it
* bank 16 signals could be moved to bank 14 and bank 16 could be depowered
* FPGA can be 50T for now (if that's cheaper and available)
* Speed grade can be -2 (1250 Mb/s serdes, 800 Mb/s DDR3 4:1)

jordens, 2019-11-09:
Leaving unused banks unpowered is apparently not recommended.

> 2) In the case of a completely unused I/O bank, leaving the VCCO pins of unused I/O banks floating reduces the level of ESD protection on these pins and the I/O pins in the bank and is generally not recommended. However, ESD events at the unconnected solder balls in the inner rows of the pin array are unlikely and not considered a high risk.
https://www.xilinx.com/support/answers/11906.html

gkasprow, 2019-11-09:
lack of bank13 complicates pin assignment and routing. Are you sure that 50T would be enough for any application?

hartytp, 2019-11-09:
My tuppence worth here is that fpgas are relatively cheap and I like to have room for expansion later. Particularly if it makes the routing easier

Having said that I’ll leave the final call to @jordens

jordens, 2019-11-09:
Can't you could use banks 14/15/16 for all the pins in 13/14/15/16 currently?

[image]

[image]

I would really like to keep the pin compatibility all the way down to 50T and lower. I'm pretty certain that 50T is fine for all applications envisioned currently.
But if you have good reasons (routing, availability, price) to go for 100T, then I'm easily convinced.

jordens, 2019-11-09:
The price difference is on the order 5-10% of the estimated board cost.

jordens, 2019-11-09:
I'm not insisting at all. Just pointing out that using bank 13 blocks using a cheaper FPGA. For the baseband application (no cost and power consumption for mixer/lo and their power supplies) that's very much noticeable. Just counting pin numbers and pin types I don't see a problem. But if you need more PCB layers because it complicates routing, then the price difference doesn't matter.
I'll probably have a number on resource consumption and gateware feasibility next week if I can get the netlist of the next RC.

gkasprow, 2019-11-09:
This means that EEM port would be splited between DAC bank (8 LVDS) and ADC bank (8 LVDS). If it is not an issue, then we can skip bank 13.
I'm a bit worried about using all pins in DAC bank due to SSO - it will be used at very high rate.

jordens, 2019-11-09:
Splitting the two EEMs between banks is fine with me as long as each EEM remains within a bank and the first LVDS pair of each EEM is on a MRCC. We plan to use one source synchronous clock on each EEM anyway.

Alternatively move the ADC LVDS stuff from bank14 into the DAC banks and use bank 14 for EEM.

Distributing the DACs evenly among banks 15 and 16 is also good. Let's try to keep it symmetric for debuggability: LVDSA and PARITYAB in one bank. And LVDSB and PARITYCD in the other bank. Distribute SYNC/DATA_CLK/ISTR/OSTR as you like among those two banks.

gkasprow, 2019-11-09:
OK, done.

closed 2019-11-09 by gkasprow

jordens, 2019-11-09:
You have DATA_CLK in each bank 15 and 16. Was that intentional?

gkasprow, 2019-11-09:
I already fixed that.