sinara-hw/Phaser #20: LO output

jordens, 2019-11-09:
[image]

I'd add the pullup R and the coupling C and u.fl on one of them.

And check https://github.com/sinara-hw/Phaser/issues/10#issuecomment-546233578 because min LO_OUT is larger than max REF_IN.

gkasprow, 2019-11-09:
@jordens look here
[image]
[image]

jordens, 2019-11-09:
That looks perfect.

jordens, 2019-11-09:
I don't think you need a matched attenuator here though. LO_OUT is open collector and EXT_VCO is high impedance. Isn't there a simple discrete match?

closed 2019-11-09 by jordens