sinara-hw/Phaser #21: baluns

jordens, 2019-11-09:
TR5A/B can be the same as TR1 and TR4. Alternatively skip them and take a single ended output of the clock fanout.

TR2/3 can also be the same as TR1 and TR4 or can be dropped alltogether (as we do on Urukul).

gkasprow, 2019-11-09:
I placed a note that explains why I used TR5A/B baluns. The clock amplitude from a single output is barely enough.
TR2/3 will be DNP by default.
TR1 and 4 are big and occupy 4x the place SOT23

jordens, 2019-11-09:
ACK then this is perfect.

jordens, 2019-11-10:
IMO the four baseband baluns TR1/4/7/6 could be changed from [ADT1-1WT](https://www.minicircuits.com/pdfs/ADT1-1WT.pdf) for significantly smaller (~1/4 area) parts, like the [TC1-1T](https://www.minicircuits.com/pdfs/TC1-1T.pdf), at the cost of 0.3 dB more loss at 300 MHz, same low f corner, reduced high-f corner from 800 down to 500 MHz.
There are also even better pin compatible versions in the pipeline for the future like the [TC1-ED9236C/1](https://www.minicircuits.com/pdfs/TC1-ED9236C-1_SPEC.pdf).

dtcallcock, 2019-11-10:
Isn't the ADT1-1WT a 75Ohm part?  Was that intentional?

jordens, 2019-11-10:
Yes. That as well. But I don't think it matters in practice.

jordens, 2019-11-10:
It comes from the [AD9910 devkit](https://github.com/sinara-hw/sinara/issues/272#issuecomment-330066162) and is used on Urukul as it plays better in the second and third Nyquist zones.

jordens, 2019-11-13:
@hartytp @dtcallcock let's settle this and #60 

I'd suggest
1. the baseband baluns to be 1:4 TC4-1Txx, with corresponding changes to the AA filter impedance for the baseband variant
2. the upconverter to have no balbals (amplitude seems right, fewer nonlinear/unmatched elements)
3. make the layout so that the upconverter could receive balbals: use the baseband balun footprint, add balbal bypass jumpers, and place the mixer bypass caps (for the baseband variant) after the balun.
4. consider using that same TC4-1Txx for TR5A/B (upconverter reference) for BOM reduction

hartytp, 2019-11-13:
I'm happy to defer to @gkasprow on this. 

> make the layout so that the upconverter could receive balbals: use the baseband balun footprint, add balbal bypass jumpers, and place the mixer bypass caps (for the baseband variant) after the balun.

Okay, but I want to keep the layout tight there to minimize parasitics/potential mismatches between the I & Q channels. Surely the methodology here is to calculate the expected output power without the balun and compare that to what the modulator wants. If we don't need a balun then better to remove it.

I'm pretty sure we'll end up with a v1.1 release pretty quickly to fix bugs, so we can always add baluns later on.

gkasprow, 2019-11-14:
I added a path with baluns for upconverter version. but if we find it difficult to route, I will remove these components.

hartytp, 2019-11-18:
I think the conclusion was to do this: https://github.com/sinara-hw/Phaser/issues/21#issuecomment-553369958

gkasprow, 2019-11-20:
OK, what did I do wrong?

jordens, 2019-11-21:
See https://github.com/sinara-hw/Phaser/issues/65#issuecomment-555157136 and https://github.com/sinara-hw/Phaser/issues/21#issuecomment-553369958 (above)

# v1.0rc3

## Baseband

[image]

## Upconverter

[image]

TR3 should be usable **in place of** R59/R2:
1. Place TR3 (DNP in upconverter) in parallel to R59/R2 (DNP in baseband)
2. Move jumpers R3 and R17 (DNP in baseband) to between DACOUTA_P/N and TR3:1/3
3. Add jumpers (upconverter DNP) between TR3:1/3 and DAC_OUT0/GND 
4. TR3 should be TC4-1T
5. Check whether that impedance ratio for baseband requires changing R11/R12/R1017/R37/R39 for that variant
6. Check whether TR3:2 or TR3:5 is the center tap and place that on the input (good for baseband)
7. Same for the other four DACs

CannaCardo, 2019-11-23:
1-4 done

gkasprow, 2019-11-29:
I can place both ADT1-1WT and TC4-1T+ as dual footprint solution.
Can we use TC4-1T for baseband?

gkasprow, 2019-11-29:
[image]

jordens, 2019-11-29:
Currently: 

[image]

I'd go for TC4-1T for baseband and bypass for upconverter. No dual footprint needed. There hasn't been any objection to that IIRC.

jordens, 2019-11-29:
But don't forget about (6) for baseband and 1:4.

hartytp, 2019-11-29:
@gkasprow once this is fixed please can you upload a new schematic rc. Hopefully we can then sign off on the schematics.

gkasprow, 2019-11-29:
Yes, I plan to do it today.

closed 2019-11-29 by gkasprow