sinara-hw/Phaser #35: EMI filter

hartytp, 2019-11-09:
@gkasprow the  EMI filter annotations imply that the DM BW is greater than the CM BW. That's not normally what one wants because the (poorly matched) DM filter will kill the InAmp's CMR (https://www.analog.com/media/en/training-seminars/tutorials/MT-096.pdf). Shouldn't we significantly increase the CM filter BW?

If so, we should check on Stabilizer/Sampler/Thermostat

hartytp, 2019-11-09:
Or, is the point that the CMC is expected to provide such good CMR at these frequencies that we're not worried? If so, we should make that clear in the annotations

gkasprow, 2019-11-13:
the CM inductance is 6.5mH and with 8.2pF it makes 689kHz filter, of course, if we ignore parasitics.
The idea was to imoprove the opamp CMRR above 100kHz
[image]

hartytp, 2019-11-14:
> The idea was to imoprove the opamp CMRR above 100kHz

That's not the only thing I'm worried about. Suppose you don't have the choke, just an RC CM/DM filter. In that case, component matching tolerances in the CM filter will lead to CM->DM conversion. This tends to dominate the overall AC CMMR, not the InAmp. As a result, one needs to add a DM filter with a much lower BW to prevent this. Without the DM filter, the AC CMMR will be much much worse than the InAmp. The DM BW should ideally be an OOM below the CM BW for this to not be an issue.

The CMC helps the situation by removing common-mode noise. But, in that case, I think you still want the CMC filter BW to be an OOM below the RC filter CM BW.

hartytp, 2019-11-14:
NB the CM bandwidth quoted on stabilizer is the LC resonance for the input CMC filter

[image]

gkasprow, 2019-11-14:
That's correct. To do this, we would have to increase the 18pF capacitor and increase the 1k8 resistor to preserve the bandwidth. Let's simulate that.

gkasprow, 2019-11-14:
Here is the schematic
[image]
And Common mode response
[image]

After adding some asymmetry to the input filter - instead of 8.2p I used 10p. Indeed, the CM gets much worse, already at 1kHz

[image]

gkasprow, 2019-11-14:
With CMC and asymmetry, it does not help much.
[image]
Even when I increase the capacitances 10x ad reduces resistance 10x it does not help much

[image]

The assymetry plays essential role.
With 2x 8,2pF (symmetrical design) it works nice
[image]
and the same, symmetrical circuit without CMC

[image]
It does help for a higher frequency, providing that the design is symmetrical.
If we want to make use of it, we would have to add trimmer and calibrate it with CM signals.

gkasprow, 2019-11-14:
If sb wants to play, here is the Tina [simulation file](https://www.dropbox.com/s/ae131ovahxb59yw/ADC_AFE.TSC?dl=0)

hartytp, 2019-11-18:
Let's leave it as is for now. The CMMR is still pretty good

closed 2019-11-18 by hartytp

jordens, 2019-11-18:
Wouldn't it be sufficient to move the common mode high-z 100k termination to before the CMC and have only differential filters after the CMC? That 100k can also be 100k to GND on negative and 1M differential FWIW. Do we need common mode RC filtering at all after the CMC?

hartytp, 2019-11-18:
@jordens I don't follow you. How is the 100k termination relevant here? 

How would having only differential filters help? 

> That 100k can also be 100k to GND on negative and 1M differential FWIW

Do you mean replacing the 100k to ground on the signal input with a 1M differential termination? We can do that, but I don't see any particular advantage.

hartytp, 2019-11-18:
> Do we need common mode RC filtering at all after the CMC?

I'm not sure I follow this. We want the differential filter to avoid rectifying pickup. The CMC doesn't do any filtering into a high-Z load, so we need the CM capacitors for it to filter. Those constraints give you the topology we have.

jordens, 2019-11-19:
That doesn't match what I've see. The CMC does change the response a lot, also just before a high-z load. We measured this a while back. I assume this is because of the interaction with the cable and the source.
And if you want a differential filter to filter (differential) pickup then you don't need common mode capacitors. As you note that kills your common mode rejection.
And third, even if you want a common mode shunt, it's simply incorrect that you are constrained to the topology here. Just split the (large) differential capacitor and shunt that center tap with a smaller cap. For f_common > f_diff this should be less sensitive to component tolerances.

hartytp, 2019-11-19:
> That doesn't match what I've seen. The CMC does change the response a lot, also just before a high-z load. We measured this a while back. I assume this is because of the interaction with the cable and the source.

To clarify: when I say "high-Z" I mean "high-Z at the relevant frequency". i.e. an open-circuited 1m BNC cable isn't high-Z by the time you get much above DC because of the ~100pF capacitance.

If we accept the model that (within the CMC BW, which is high in our case) a CMC is a small differential-mode inductance and a large common-mode inductance, it can't do anything if the load is truly high-Z (by which, I mean that at the relevant frequency it's impedance is high compared with the CMC inductance).

The InAmp input capacitances are only ~10% of the capacitances used in the CM/DM filters we use. So, if we scrapped the CM filter capacitors, we'd move the CM filter pole up by an order of magnitude, which would significantly reduce the CM filtering.

[image]

[image]

When you say this doesn't match what you've seen, do you mean looking at Zotino's outputs on a scope? If so, the cable + scope capacitance stop being a high-Z load at relatively low frequencies.

hartytp, 2019-11-19:
> And if you want a differential filter to filter (differential) pickup then you don't need common mode capacitors. As you note that kills your common mode rejection.

No, that's not what I'm saying. The decreases common-mode rejection at AC comes from two places: the mismatch in the resistors, which are common to both the CM and DM filters, and the mismatch in the DM filter capacitors.

It's relatively cheap to use 0.1% resistors for the DM filter, but hard to get the capacitors better than 5%. As a result, the dominant source of DM->CM conversion is typically the capacitors. This is why the standard design for EMI filters (see the app note I linked above) has a CM filter with a cut-off below the DM filter; the extra CM filter mitigates the loss in CMMR caused by the DM filter. i.e. if you have a DM filter then the CM capacitor will improve the AC CMMR, not reduce it.

hartytp, 2019-11-19:
Nope, sorry, that's not a correct explanation. See here for a better explanation: https://www.analog.com/media/en/technical-documentation/application-notes/AN-671.pdf

The point is this:
- even with the CMC you still need a the two CM filter capacitors to prevent rectification of CM noise. If you don't use external capacitors then you're relying on the parasitic capacitances, which isn't really better
- you also want a DM filter to prevent rectification of DM noise
- the DM filter cut-off should generally be lower than the CM filter cut-off to make sure we catch CM->DM conversion from the CM filter

hartytp, 2019-11-19:
NB in the app note I linked to they do have a design with only a CMC: 

[image]

However, as I understand it:
(a) This doesn't do any DM filtering, which I would like to
(b) it relyies on forming a filter between the parasitic (e.g. OpAmp input) capacitance and the CMC inductance. For the choice of CMC and InAmp we have, that filter cut-off would be a bit too high.

jordens, 2019-11-19:
> To clarify: when I say "high-Z" I mean "high-Z at the relevant frequency". i.e. an open-circuited 1m BNC cable isn't high-Z by the time you get much above DC because of the ~100pF capacitance.

That's correct. But it's not the situation that I measured. I had a low-z source-CMC-coax-CMC-high-z load setup and having the second CMC made a big difference. It's the same case here. Between the long coax and the high-z INA input the CMC will make a difference (at AC).

And regarding the filter setup: read again what I described. It's the standard wye-delta transform. It makes a difference in component tolerance sensitivity if the CM and DM impedances are different. I am pretty sure that faced with mismatch, a Y/T termination/filter does less CM to DM conversion than a delta/pi.

hartytp, 2019-11-19:
> And third, even if you want a common mode shunt, it's simply incorrect that you are constrained to the topology here. Just split the (large) differential capacitor and shunt that center tap with a smaller cap. For f_common > f_diff this should be less sensitive to component tolerances.

hmm...I haven't seen that topology used before.  From a quick think, I think that would help in the limit where:
- the resistors are tight tolerance (say 0.1% or better)
- the DM capacitance is >> than the CM capacitance
- we can ignore parasitics like the OpAmp input/diode/trace/etc capacitance (or, at least assume that they are very well matched between the two channels)

We'd need to simulate to figure out how much of a difference it would make in practice for our circuit. But, e.g. if we split the DM capacitor in two it's only about twice the size of the CM capacitors. Similarly, the total parasitics are probably only a factor of a few lower than the CM filter capacitors. So, not sure how much it would actually change things, but I agree it's another option that I hadn't thought of.

hartytp, 2019-11-19:
> That's correct. But it's not the situation that I measured. I had a low-z source-CMC-coax-CMC-high-z load setup and having the second CMC made a big difference. It's the same case here. Between the long coax and the high-z INA input the CMC will make a difference (at AC).

At what frequency? The only way I understand that is if the high-Z load setup wasn't high-Z compared with the CMC inductance.

Anyway, as I said, I agree that you can build a filter where the CMC capacitance is dominated by the parasitics, but I don't see a reason why it would be a good idea to do that in our case.

jordens, 2019-11-19:
That was a 10X 1M scope probe at <MHz IIRC. https://github.com/sinara-hw/Zotino/issues/21#issuecomment-376119235