sinara-hw/Phaser #38: RC components

gkasprow, 2019-11-10:
For such a wide-band circuit, we cannot use standard components.
Most capacitors are not capacitors anymore at 4GHz.
That's why to have a frequency range from 1 to 4GHz we need to use either different components or connect them in parallel.
For example, 1uF,16V, 0402 from Murata (GRM033R61A105ME44)
[image]
Its impedance is still a few ohms but has an inductive character. It's hard to predict how it won't make resonance with parasitic capacitances.
Anyway, I will do a simulation with S models of all RF components in the path to see if we won't have any unwanted effects.

hartytp, 2019-11-10:
Can’t we use different bom options with the same pads for baseband 1MHz to 400MHz) and upconverting 400MHz to 4GHz?

gkasprow, 2019-11-10:
yes, we can.

jordens, 2019-11-10:
Also #39 

I'd use different capacitors for the two variants.

dtcallcock, 2019-11-10:
Or you can use a something like an [ATC530L}(https://www.atceramics.com/UserFiles/530l.pdf), which is specifically designed for DC blocking over many octaves without these issues.  They are $0.4 ea so BOM variant option will be cheaper, but there may be some value in reducing the number of components that need switching.

jordens, 2019-11-10:
IMO it's unrealistic that those capacitors will be the right cure for a problem we haven't quantified. For the upconverter there is only one decade. Just choosing a capacitor value to place the ESR in that band will be good enough (and making sure that the capacitors are not special high-Q is pretty close to a magic (i.e. lossy) broadband capacitor). It's easier to give it a test on Mirny to see whether there are any problems with this circuit and 10 nF blocks.
The baseband version is from 1-400 MHz. Same as Urukul where we run with 10 nF throughout and see no dramatic problems. If we take the usual 100 nF here in a small package, we'll be fine.

dtcallcock, 2019-11-10:
> The baseband version is from 1-400 MHz. Same as Urukul where we run with 10 nF throughout and see no dramatic problems.

Had the amplitude vs. frequency of urukul been carefully measured by anyone yet?

jordens, 2019-11-10:
I looked at it for low frequencies (1-5 MHz) and was happy enough. IIRC we chose the corner to be around 1 MHz.

gkasprow, 2019-11-13:
Once we find some issues, I will add assembly variant with other coupling capacitor values.

hartytp, 2019-11-18:
The upconverting variant still has 1uF caps. @gkasprow please ensure that the entire RF signal path has appropriate DC block caps in both variants (cf https://github.com/sinara-hw/Phaser/issues/39).

gkasprow, 2019-11-20:
True. I will add the variant once @CannaCardo finishes the layout. In Altium one can use Variants to modify Comment field. This makes the BOM messy, so I place 2 components footprints one on top of another and assemble one of them. This is not very elegant but works.

gkasprow, 2019-11-29:
I added two sets of caps - 10nF and 1uF to support both assembly variants. The footprints are placed one on top of another.

closed 2019-11-29 by gkasprow