sinara-hw/Phaser #41: power supplies

hartytp, 2019-11-11:
@gkasprow I'm fine  with the current supplies, but a little curious why we've departed from models used in other recent designs...
- Why are we using the Tracoo SMPS instead of the TDK one (lower current/smaller package of same family)? Didn't we have issues with those ones breaking on the DIO cards?
- Why are we using the LT1761/TL1864 instead of the 3042 etc? Again, I don't really object here, but I just like to use as many ICs as possible that we've used in other well-tested designs (didn't these ones have oscillation issues?)

If you're happy with the current design just close this

gkasprow, 2019-11-13:
This TRACo SMPS is much smaller than TDK. We used other types on the DIO cards. 
I use LDOs that have limit current lower than SMPS. Of course, we can add current limit resistor to the 3042.
 LT1761/LT1864 doesn't need compensation caps and we used them already and is much simpler than the solution used in Fastino.

hartytp, 2019-11-13:
Okay, let's give it a go then

closed 2019-11-13 by hartytp