sinara-hw/Phaser #44: Clock fanout power

hartytp, 2019-11-11:
Shouldn't this be powered from an LDO to avoid getting large SMPS spikes on the DAC output?

hartytp, 2019-11-11:
NB this should ideally power on after the DAC 1V2 supply to avoid sequencing issues

hartytp, 2019-11-11:
Also, are you happy with the power sequencing of the fanout buffer supply relative to the modulator/DAC power supplies?

gkasprow, 2019-11-13:
The clock buffer supply (P1V8) wakes up after DAC P1V2A and modulator P3V3A   so it is fine.

hartytp, 2019-11-13:
> The clock buffer supply (P1V8) wakes up after DAC P1V2A and modulator P3V3A so it is fine.

Yes, you're right. So, no sequencing issue. However, I'm still a little concerned about running this from a SMPS. It's low power so can't we just use a LDO?

hartytp, 2019-11-13:
> NB this should ideally power on after the DAC 1V2 supply to avoid sequencing issues

I think I just got confused between the 1V2 and 1V2A rails. As you say, the sequencing looks correct.

gkasprow, 2019-11-13:
OK, let's make the 1V8A rail then out of 3.3V

closed 2019-11-13 by gkasprow