sinara-hw/Phaser #47: DAC CLKVDD

hartytp, 2019-11-11:
Is a ferrite sufficient isolation between the (noise sensitive) CLKVDD supply and the DACVDD/DIGVDD supplies? Should this have an LDO?

hartytp, 2019-11-11:
Simillarly, for DACVDD (noise sensitive) and DIFVDD (digital noisy rail)

hartytp, 2019-11-11:
On the eval board it looks like the clk and analog supplies come from an LDO and are isolated by a ferrite, while the digital supply comes directly from a SMPS. Worth copying that? 

[image]

hartytp, 2019-11-11:
same goes for running the IO rails from the same supply as analog rails. Surely better to run the LVDS rail directly from an SMPS rather than running it from the sensitive analog rail???

hartytp, 2019-11-11:
Equally, are we happy with the decoupling between the DAC and modulator supplies?

gkasprow, 2019-11-13:
All the specs in the datasheet are usually taken with eval kit, so if we are happy with them, I'd simply copy that. And this is what I did. I can add a dedicated LDOs for digital signals if this makes you sleep better :)
In the case of the modulator, we are mainly affected by RF. I used 1k Ferrite beads as suggested in the DS.
Again, I can replace this big LDO with 3 smaller ones if we need better separation, but I don't think that it would be an issue. More important is how we route the signals on the PCB.

hartytp, 2019-11-13:
> All the specs in the datasheet are usually taken with eval kit, so if we are happy with them, I'd simply copy that. And this is what I did. 

I don't think so, see my post above.

This part of the eval schematic shows that the DAC analog rails are generated from LDOs, while the digital parts are generated from a SMPS.

There are then additional ferrites used for decoupling.

> Again, I can replace this big LDO with 3 smaller ones if we need better separation, but I don't think that it would be an issue. More important is how we route the signals on the PCB.

Okay, that makes sense.

hartytp, 2019-11-18:
@gkasprow the design still does not follow the eval board. Look at the image below. The DAC digital supplies come direct from a SMPS while the analog supplies come from a dedicated LDO. This allows them to be well decoupled from the digital. In contrast, we just use a FB.

[image]

gkasprow, 2019-11-20:
OK, we can then connect it to P1V2 which is also generated by an LDO and is used to supply the FPGA.
We can also add another LDO, dedicated to digital part of DACs.

hartytp, 2019-11-23:
@gkasprow we should fix this before routing as it affects power planes

gkasprow, 2019-11-28:
I copied P1V2A LDO and connected it to DIGVDD. IT should solve the issue.
[image]

hartytp, 2019-11-28:
Pic didn’t upload :( isn’t it better to drive from a smps to avoid significant power dissipation in the ldo?

gkasprow, 2019-11-29:
we don't have 1V2 SMPS rail.
[image]

gkasprow, 2019-11-29:
anyway, 300mV drop * 600mA = 180mW.

hartytp, 2019-11-29:
I was suggesting adding a new rail but I agree if the ldo drop out is really that low it’s fine

closed 2019-11-29 by hartytp