sinara-hw/Phaser #49: PLL LPF

hartytp, 2019-11-11:
@WeiDaZhang can you check you're happy with the current DAC PLL loop filter please?

[image]

hartytp, 2019-11-11:
Same for the modulator LPF

[image]

WeiDaZhang, 2019-11-11:
There is not much information provided about the loop filter, apart from a paragraph at top page.4
https://www.ti.com/lit/pdf/SLAA520?jktype=tech_docs
The suggestion would be:

1. Not to use the on-chip PLL, unless one has to do so.

2. Not to make the divide ratio greater than 120 when using the on-chip PLL, and therefore avoid external loop filter, unless one has to do so.

3. Use the default loop filter setting which is as the schematic is.

hartytp, 2019-11-11:
> Not to use the on-chip PLL, unless one has to do so.

Why? From the DAC data sheet, it seems pretty nice.

> Not to make the divide ratio greater than 120 when using the on-chip PLL, and therefore avoid external loop filter, unless one has to do so.

I don't follow? Why is the divider connected to using an external loop filter? In our case, the DAC clock will be 1GHz with 125MHz ref (max f_pfd is 155MHz according to data sheet). 

> Use the default loop filter setting which is as the schematic is.

Okay, let's do that then...

jordens, 2019-11-11:
Because there is already an internal LF

[image]

But please add a 0R and a bit of space to that pin:

[image]

WeiDaZhang, 2019-11-11:
> > Not to use the on-chip PLL, unless one has to do so.
>
> Why? From the DAC data sheet, it seems pretty nice.]

It was said in the document, however, taking a closer look.
[image]
To put it in a graph we had before, it seems quite good, you are right.
[image]

hartytp, 2019-11-11:
Thanks @jordens @WeiDaZhang 

@WeiDaZhang what about the upconverter LF? are you happy with that?

gkasprow, 2019-11-13:
>But please add a 0R (DNP) and a bit of space to that pin:

@jordens do you want to have the LF disabled by default?

jordens, 2019-11-13:
Yes. We are looking at P*M < 64 for 125 MHz and < 80 for 100 MHz reference in. 10 MHz wouldn't work anyway.

hartytp, 2019-11-18:
@WeiDaZhang please can you confirm that you're happy with the upconverter LF asap and then close this? Thanks!

WeiDaZhang, 2019-11-18:
I'm looking at it. What's the frequency PFD? I suppose it's 125MHz?

WeiDaZhang, 2019-11-18:
@hartytp

jordens, 2019-11-18:
62.5 MHz

WeiDaZhang, 2019-11-18:
This one should work. I've tried for two frequencies.
The VCO gain is not clear, I assumed (4800-2400)/5 = GVCO (MHz/V) when VCO range DIV = 1 selected. If the tuning voltage is actually 3.3V, there will be a loop bandwidth error but should remain stable.
I've allowed relatively wide loop bandwidth assuming the reference is much lower noise than the VCO even with frequency scaled.
[image]
[image]
C1       C2        R2    C3    R3    C4    R4
120p   3000p  330   33p  330  33p  330

hartytp, 2019-11-18:
@gkasprow can we go for @WeiDaZhang's loop filter please? Once the boards arrive, we'll test this on a phase noise meter and optimise the loop filter further.

gkasprow, 2019-11-28:
done
[image]

closed 2019-11-28 by gkasprow

pathfinder49, 2021-03-29:
> > > Not to use the on-chip PLL, unless one has to do so.
> > 
> > 
> > Why? From the DAC data sheet, it seems pretty nice.]
> 
> It was said in the document, however, taking a closer look.
> [image]
> To put it in a graph we had before, it seems quite good, you are right.
> [image]

@hartytp What's the conclusion here? Is the idea to use the DAC PLL or bypass it?

hartytp, 2021-03-29:
> @hartytp What's the conclusion here? Is the idea to use the DAC PLL or bypass it?

Depends on the application.

Using the DAC/upconverter PLL and clocking from Kasli is generally the most convenient way of setting things up. In many cases that will be completely fine from a noise perspective.

If one needs the lowest possible noise then an external clock/LO will definitely give better results than using an internal PLL from the DAC / upconverter. NB the DAC clock is generally much less important than the upconverter (if it's being used) since, all things equal, phase noise scales like `20*log10(frequency)` so for a 3GHz carrier and 300MHz IF, one is 20dB more sensitive to the upconverter LO.

My advice would be to measure the phase noise for the upconverted RF and then decide if it's good enough for the gate. Using an external microwave source for the LO / DAC clock is there are a fallback plan if we decide it's needed.