sinara-hw/Phaser #54: modulator input termination

hartytp, 2019-11-11:
Don't the modulator baseband inputs need external termination (see eval board)?

hartytp, 2019-11-11:
[image]

jordens, 2019-11-11:
They have termination just like in the datasheet.

[image]

closed 2019-11-11 by hartytp

gkasprow, 2019-11-13:
We should place this termination circuit after the 0R option selectrion resistors. I will move them to the mixer page.

hartytp, 2019-11-13:
okay, just keep the layout as tight as possible to minimize parasitics and make sure there is very close matching between the I and Q channels