sinara-hw/Phaser #56: modulator ref coupling capacitor

hartytp, 2019-11-11:
Isn't 4p7 a bit small for a 125MHz ref clk?

[image]

hartytp, 2019-11-11:
sorry, this isn't for a reference. Still, any reason to use such a small capacitor here?

gkasprow, 2019-11-13:
it was used in the devkit so I copied it.

jordens, 2019-11-13:
it's a high impedance input, probably the same 4-5k || a few pF as the other inputs.

hartytp, 2019-11-13:
okay, true, in that case let's leave it as is

hartytp, 2019-11-13:
hmm...I still would have thought we'd want to the coupling capacitance a bit higher. Otherwise we have a significant divider ratio with the input capacitance if it's a few pF. Is there any downside in using 100pF?

gkasprow, 2019-11-13:
OK, let's go for 100p

closed 2019-11-13 by gkasprow

jordens, 2019-11-13:
The downside to large coupling capacitors is worse low-f rejection (power supplies on these single-ended lines) which becomes phase noise once it hits the input buffers. Let's leave it for tweaking later.