sinara-hw/Phaser #57: FP BB/UC LED

hartytp, 2019-11-11:
To indicate which variant is populated

jordens, 2019-11-11:
Can't we just put some marking/label on the panel? Using a LED for something that's guaranteed to not change is bad. I'd rather use that one remaining LED to signal external clock input selection. The other LEDs are all channel-associated.

hartytp, 2019-11-11:
Okay, I'm fine with adding some text to the panels (separate LEDs felt easier than different panels). So long as there is a clear external marking.

jordens, 2019-11-11:
Alternatively we can sacrifice one of the 8 channel LEDs to talk about variants. E.g. red for baseband and blinking red for Mixer PLL unlocked.

gkasprow, 2019-11-13:
we will have other FPGA bits anyway.
I will make two panel designs as we have for Urukul.

gkasprow, 2019-11-14:
There is a variant signal routed to FPGA. It can be forwarded to the LED. So we can have both panel annotation and LED.

closed 2019-11-14 by gkasprow