sinara-hw/Phaser #59: EXT VCO attenuator

hartytp, 2019-11-11:
I couldn't see where in the data sheet it says we need this: 
[image]

I did see that -10dBm was the value they happen to use in some plots, but I didn't see any indication that this is a requirement...

hartytp, 2019-11-11:
Also, this should be DNP in the BB variant. FWIW I would have thought we could do this with standard resistors since it doesn't need to be precise...

gkasprow, 2019-11-13:
It depends on what frequency we will pass thgrough this path.
OK, let's add resistive divider.

hartytp, 2019-11-13:
@gkasprow if you think that 4GHz through a resistive divider is problematic then please do keep the attenuator (assuming we need it but, as I said, I didn't find a spec for the LO power the modulators want).

gkasprow, 2019-11-13:
this could be problematic for the standard resistors. But since we don't know if we need it, lets switc hto resistors anyway.

hartytp, 2019-11-13:
hmm...before closing this it would be better to decide if the attenuator is actually needed.

Let's leave the packaged attenuator for now and an action on me to figure out what LO power is wanted.

gkasprow, 2019-11-13:
we can always place one with other attenuation. 0R 0402 resistor can be soldered as well instead of it.

hartytp, 2019-11-18:
@gkasprow was the plan to put a pad 0402 resistor that can be used to bypass this attenuator?

gkasprow, 2019-11-20:
Yes. it can be soldered directly to the attenuator pads.

gkasprow, 2019-11-28:
I added 0R to make sure it can be placed by P&P machine

closed 2019-11-28 by gkasprow