sinara-hw/Phaser #60: baseband amplitude

jordens, 2019-11-11:
Currently the path between the DAC and the mixer leads to some 0.5 V (25 Ohm, 20 mA) full scale current (peak single ended). The mixer has optimal performance (OIP3 e.g.) around 0.5 V single ended RMS, i.e. 0.71 V peak single ended.

Proposal: use a 1:4 transformer after the AA filter. That space is needed anyway for the baseband variant baluns. It gives more power at the output in both variants. There is a coarse full scale adjustment to fine tune it further. Also the eval board uses the 1:4 transformer.

hartytp, 2019-11-11:
> 20 mA

Isn't the DAC 30mA FS?

jordens, 2019-11-11:
20 mA typ.

hartytp, 2019-11-11:
> 20 mA typ.

IIRC there is a resistor one picks dependent on the FS current and the one we used is for 30mA. Maybe I'm wrong though...

jordens, 2019-11-11:
You'd operate it with `coarse_dac` setting it to 20 mA FS.

hartytp, 2019-11-11:
Okay, but should we change `BIASJ` then?

jordens, 2019-11-11:
No. It's the product of both.

hartytp, 2019-11-11:
okay, thanks for clarifying.

Anyway, this sounds like a good idea.

What's the 1:1 balun for?

hartytp, 2019-11-11:
@dtcallcock what's the actual min frequency you want? 1MHz or lower?

It looks like that 1:4 should be fine though

[image]

jordens, 2019-11-11:
I guess that two-transformer combo blocks common mode effects (e.g. the primary secondary capacitive coupling which is an intrinsic limitation in single-core baluns) a bit better. Don't know any numbers though.

jordens, 2019-11-11:
On the other hand the mixer datasheet talks about "VinBB = 300 mVrms single-ended sine wave in quadrature" as their favorite characterization default. So maybe 500 mV peak SE per quadrature is just right.

hartytp, 2019-11-11:
It does look like the mixer is designed to connect to a DAC without a balun:

[image]

Unless I'm missing something, DAC3283 looks very similar to 34H84 with regards to outputs...

hartytp, 2019-11-11:
No, never mind, it is a bit different

dtcallcock, 2019-11-11:
> @dtcallcock what's the actual min frequency you want? 1MHz or lower?

1 MHz should be good but it'd be nice to have at least somewhat graceful degradation down to around 300kHz if possible.

There are better and smaller 1:4 baluns for the lower frequency end (eg. [TC4-1TX+](https://www.minicircuits.com/pdfs/TC4-1TX+.pdf).  What upper frequency are you shooting for with the BB variant though?

hartytp, 2019-11-11:
> There are better and smaller 1:4 baluns for the lower frequency end (eg. TC4-1TX+. What upper frequency are you shooting for with the BB variant though?

This is just on the DAC, so I think the choice here is agnostic of which variant we're using; both need to go up to around 400MHz.

dtcallcock, 2019-11-11:
It says the TC4-1TX+ is useable to 450MHz but they don't spec it above 300MHz.  The [TC4-6TX+](https://www.minicircuits.com/pdfs/TC4-6TX+.pdf) has worse performance but the return loss holds up better above 200MHz and it's specified up to 600MHz.

jordens, 2019-11-11:
The current AA filter is already 10 dB down at 400 MHz. I wouldn't worry about the balun adding another ~3 dB.

gkasprow, 2019-11-13:
I assume we leave it as it is now.

closed 2019-11-14 by gkasprow