sinara-hw/Phaser #65: layout

gkasprow, 2019-11-14:
@CannaCardo it's your turn with layout and routing
- copy stack up from Kasli, the same with rules
- copy position of IDC connector and assembly holes from Urukul
- make component placement, once finished post it for review
- then during routing follow strictly reference design for ADC and upconverter

gkasprow, 2019-11-14:
also:
- add missing diff nets and directives
- if you encounter any difficulties, open an issue
- if you have some questions, want propose some changes, open issue
- if you find some bug, open issue.

jordens, 2019-11-15:
I think a couple of issues were addressed incorrectly. Like the ufl for the baseband q output, the baseband balun type, the upconverter balbal option and bypass jumper design, the input shield grounding.

hartytp, 2019-11-18:
> Like the ufl for the baseband q output

What's the issue here? Are you not ok with MMCX?

hartytp, 2019-11-18:
> bypass jumper design

What do you want done differently here?

hartytp, 2019-11-18:
@jordens I've done a second review focussing on the items we raised previously and opened issues where necessary. Did you spot anything I've missed?

jordens, 2019-11-18:
The baseband q output doesn't need to be MMCX. It's very much debugging or prototyping only. MMCX is much bigger and pierces though all layers. Baseband Q would be disabled by default and should be DNP in later variants.

The bypass jumper should be as discussed elsewhere. It is not. There should be the same transformer footprint, populated with a single ended secondary for baseband or DNPed (but optionally a 1:4 or 1:2 to increase amplitude) and differential output plus differential termination into the mixer for uconverter.

I had spotted the issues I mentioned above.

hartytp, 2019-11-18:
> The bypass jumper should be as discussed elsewhere. It is not. There should be the same transformer footprint, populated with a single ended secondary for baseband or DNPed (but optionally a 1:4 or [...]

Is this all covered by https://github.com/sinara-hw/Phaser/issues/21 ?

jordens, 2019-11-18:
yes.

jordens, 2019-11-18:
Check out http://jordens.github.io/sandbox/ by the way. Smaller than the PDFs, faster on mobile, nice for layouts. If someone wants to extend and manage this for to all sinara-hw projects feel free.

gkasprow, 2019-11-20:
I found u.fl connectors very fragile and treat them only for debugging.
I can use SMT MMCX if you want to use them more than once:)

gkasprow, 2019-11-20:
@jordens is there a way one can generate automatically a link for the design? I'd love to place such one in wiki of every project.

jordens, 2019-11-21:
Sure. This can be automated and maintained. Somebody would have to do it though.

jordens, 2019-11-21:
Ok. You choose the connector. Of it works I don't mind either way.

hartytp, 2019-11-21:
So long as there is space, I definitely prefer MMCX for this.

closed 2019-11-23 by hartytp