sinara-hw/Phaser #73: Layout

gkasprow, 2019-11-23:
- [ ] swap J9 and SATA connectors. Otherwise CPCIS adapter won't have access to the J9. One can swap transceivers if needed.
- [ ] separate channels with L28, L36 etc . The inductors may couple and cause crosstalk
- [ ] U can swap IC5 outputs

gkasprow, 2019-11-23:
- [ ] do not place vias in pads. U can switch to 0201 caps under FPGA to fit them between vias. make sure that via hole does not touch the component pad. Otherwise, it will suck the tin.
- [ ] don't even try routing on 6 layers. Move to 8 or copy the stackup from Kasli. Use the Brandner web tool for stackup creation.

gkasprow, 2019-11-23:
- [ ] move LM75 closer to the heat source: ADC and upconverter

jordens, 2019-11-23:
I think kasli is 6 layers. Would be nice to not have to go beyond kasli.

gkasprow, 2019-11-23:
Kasli is purely digital board. It has only one ground plane. Doing Phaser with only 6 layers would be tricky.

gkasprow, 2019-11-23:
Anyway, let's try to copy the DAC layout/routing from evkit and see how many layers it needs.

hartytp, 2019-11-23:
@gkasprow can you finish off the remaining schematic issues please?

hartytp, 2019-11-23:
@CannaCardo good work. Can we have the top/bottom side renders in the PDF as well please?

CannaCardo, 2019-11-23:
Renders added

hartytp, 2019-11-23:
Thanks! Do you think we can get the boards off to manufacture this week?

CannaCardo, 2019-11-23:
Most likely :)

hartytp, 2019-11-23:
Amazing, THANK YOU!

dtcallcock, 2019-11-24:
First 4 SMAs are labeled RF OUT 0-3 on silkscreen.

hartytp, 2019-11-29:
how's the layout going @CannaCardo ?

gkasprow, 2019-11-29:
I took over the design to implemented all fixes. Then I will pass it to @CannaCardo to finish. I should finish them today.

hartytp, 2019-11-29:
Aah, ok, makes sense! Thanks! (I have too many projects on the go and am trying hard to get them finished)

gkasprow, 2019-11-29:
well, at the moment I have 17 projects going. Sinara is one of them. Some of them started 10years ago and don't want to finish...

gkasprow, 2019-11-29:
@CannaCardo during finishing the routing make sure that:
- [ ] no components are placed close to the FPGA to enable assembly of the heatsink latch
- [ ] follow precisely routing of the upconverter chip Now the decoupling capacitors are far from the IC pins. The inductors or ferrite beads can be on the bottom layer or far away but the caps must follow the datasheet recommendation. Lower capacitors values must be installed closer to the IC
- [ ] All QFNs have vias in the thermal pad
- [ ] all test point, LEDs and connectors have labels which indicate their function
- [ ] all ICs, diodes, connectors, and inductors, transformers have their designator enabled. Use 1mm high font, 0.2mm stroke
- [ ] all vias in thermal pads are unmasked on the opposite side to prevent from acid pocket formation
- [ ] there are at least 2 different fiducials per side on top and bottom. Use small ones.

hartytp, 2019-11-29:
Thanks @gkasprow ! Can you post a new rc so we can sign off on the schematic please?

gkasprow, 2019-11-29:
one moment.

gkasprow, 2019-11-29:
the latest release is here
https://github.com/sinara-hw/Phaser/releases/tag/v1.0rc5
@CannaCardo you can take over the project and finish the routing.

jordens, 2019-11-30:
And put exposed polygons on the bottom layer under the two mixers like we have for the DDS on Urukul.

jordens, 2019-11-30:
And make sure that there are no tall components (that big capacitor) on the bottom that would interfere with card insertion.

jordens, 2019-11-30:
And Greg's point from above is not honored. You really need to spread out the signal paths after the DACs, especially the inductors. Even within a single diff pair the coupling between the two inductors is problematic as it reduces the filtering.

jordens, 2019-11-30:
And PI simulation. There are some really carved up polygons.

hartytp, 2019-12-01:
AFAICT all schematic review items were correctly implemented. Good work!

Layout is coming along nicely @CannaCardo! Really looking forward to getting my hands on phaser. Good work all.

hartytp, 2019-12-01:
@jordens assuming you have no objections, I suggest we freeze the schematic here.

hartytp, 2019-12-07:
@CannaCardo whats the status of this? Are we on track to get the boards to manufacture next week?

CannaCardo, 2019-12-07:
I've got almost all of the issues here done, few traces with length matching (mainly DAC area) and PI simulation are left.

hartytp, 2019-12-07:
Ace :)

So ready for review early next week sounds about right then? That would be wonderful

CannaCardo, 2019-12-07:
Yes. I've finished other projects, so there are no excuses now  :)

hartytp, 2019-12-09:
@gkasprow can you do a layout review?

dtcallcock, 2019-12-09:
Just noticed that R49 is still populated on baseband version.

gkasprow, 2019-12-09:
I finished my review

hartytp, 2019-12-09:
@gkasprow @CannaCardo a few questions:
-are you satisfied that all converters follow eval board / data sheet recommendations for the layout?
- are you happy with the digital to analog isolation and isolation of the analog from the fpga + converters?
- are the I/q channels perfectly matched? We need to avoid any asymmetry to keep the sideband suppression good
- happy with Si for high speed and analog traces?

gkasprow, 2019-12-09:
@CannaCardo, please mark remaining "ticks" above.

hartytp, 2019-12-09:
- happy with layout and routing of the rf signals to give good si and minimise parasitical in filters etc

gkasprow, 2019-12-09:
Once all the issues I posted are addressed, I would be happy.
In the end, I will make a few GND cutouts to limit the spreading of EMI from DC/DC to RF part.

hartytp, 2019-12-09:
Copying the only unchecked issue from above:

- [ ] follow precisely routing of the upconverter chip Now the decoupling capacitors are far from the IC pins. The inductors or ferrite beads can be on the bottom layer or far away but the caps must follow the datasheet recommendation. Lower capacitors values must be installed closer to the IC

hartytp, 2019-12-09:
Anyway, very excited that we're now so close to getting this off. Just Booster after this...

hartytp, 2019-12-09:
One really nit picky thing that you may feel isn't worth it...
- [ ] on stabilier all the power TPs and LEDs are in a row with a couple of GND pads there as well. That made spotting mistakes and checking supplies really quick (don't need to search for them and you can really see if one supply isn't coming on properly). If it's not too much work we should to that here as well

CannaCardo, 2019-12-11:
>  on stabilier all the power TPs and LEDs are in a row with a couple of GND pads there as well.

I've added GND pads visibly near the testpoints, should be easy to find

hartytp, 2019-12-11:
What was nice on Stabilizer was that all TPs and LEDs were in a row, so one could check all power rails by eye really quickly without having to search over the board. That's much nicer than having them scattered over the board. But, I understand if it's too much work to do that now.

CannaCardo, 2019-12-11:
I'll keep that in mind with future boards ;)

hartytp, 2019-12-13:
@CannaCardo how's this going? Any chance of getting another RC today so we can do our final reviews over the weekend and send off to production on Monday?

CannaCardo, 2019-12-13:
I added the next RC, still without PI/ copper balancing as I don't have access to a hyperlynx license at the moment.

gkasprow, 2019-12-13:
I will simulate it with Ansys PDN or CST PDN over the weekend. I have a private license for both.

hartytp, 2019-12-13:
@CannaCardo awesome! Did you folow the upconverter reference design layout? If so, I think all issues here are resolved and we can close this. Well done.

CannaCardo, 2019-12-13:
Yes, the image on the schematic is upside-down compared to the PCB

hartytp, 2019-12-13:
Great, thanks!

gkasprow, 2019-12-13:
yes, layout is OK, but I still don't think the ref design is done correctly:)

hartytp, 2019-12-13:
@gkasprow are you happy with the layout? In particular the usual things like:
- loop area in high-frequency/low-noise (LF etc) signals?
- position of decoupling capacitors
- IQ and channel-channel matching/symmetry
- noise sources (SMPSs/digital traces)
- controlled impedances

etc? If so let's close this and send to manufacture after PI

hartytp, 2019-12-16:
@gkasprow thanks for doing all that!

It looks like most open issues can be closed now. What’s left to do? Can you post a final rc once all changes are complete so we can sign off?

closed 2019-12-17 by gkasprow