sinara-hw/Phaser #74: Mmcx ref clock input location

jordens, 2019-11-23:
Should be as far back as possible. And the cable should not interfere with the eem ribbons.
The location of the mmcx on urukul and mirny is not very good.

Jtag on the other hand can be pretty much anywhere. Be need for it to be in the back.

CannaCardo, 2019-11-25:
Would here be ok, considering the CPCIS adapter?
[image]

jordens, 2019-11-26:
The current cPCI-S design has a specific MMCX clock output that derives it from a diff pair on the cPCI backplane. But it doesn't have that gigabit serial connector yet.
I think for c-PCI-S the clock should be placed on the gigabit connector and there should be no MMCX jumper. To do that (a) place MMCX in just above the right EEM, (b) the gigabit connector above the left EEM (or between the EEMs if that fits), (c) add jumper capacitors to switch the clock between the balun and a pair on the gigabit connector. @gkasprow does this make sense?

gkasprow, 2019-11-28:
yes. done.

closed 2019-11-28 by gkasprow

gkasprow, 2019-11-28:
[image]

gkasprow, 2019-11-29:
btw, recent CERN DI/OT board based on 500$ ZU7 SoC has also gigabit line per CPCIS slot + 15 LVDS + low jitter clock. WIth one assembly option the user can decide if one needs 16-th LVDS or gigabit link.
The project is here https://ohwr.org/project/diot/wikis/home
In a few days I will update schematics.

gkasprow, 2019-11-29:
the Open source CPCIS backplane is already designed (in Kicad) and goes for production.
Low-cost 400W PSU is in the design phase.