sinara-hw/Phaser #78: output transformer

hartytp, 2019-12-04:
https://github.com/sinara-hw/Urukul/issues/39#issuecomment-561565760

jordens, 2019-12-04:
This is already done. The baseband version is supposed to have that 4:1 balun. (nb I don't know whether anybody did what I asked and cross-checked that the diff filter works fine at that impedance).

hartytp, 2019-12-04:
Okay. I suggest we leave this for now while @CannaCardo does the layout, but remember to check it before production.

gkasprow, 2019-12-09:
@hartytp do you want us to add the power amplifier at the output?

hartytp, 2019-12-09:
No let’s leave it as is for now

hartytp, 2019-12-09:
@gkasprow what do you think about balun selection for phaser?

gkasprow, 2019-12-09:
To answer your question we should make a simulation with MO or DesignerRF. Since it is a single element and it is easy to replace, we will simply do so if the performance of the existing one is not sufficient. We do not use baluns for IQ operation so their mismatch is not an issue here.

hartytp, 2019-12-09:
It would definitely be nice to use a balun that gave a bit more output power since we're not using a pre-amp.

hartytp, 2019-12-09:
Also, I'm not totally clear on what measurements we would want to make to verify the performance in the current arrangement.

gkasprow, 2019-12-09:
I'd measure THD and power for interesting frequencies.

hartytp, 2019-12-09:
ok, will do. That's easy to do and was on the list any way.

hartytp, 2019-12-09:
Sorry, looking again at the comments above I realise that there isn't anything to do here.

The only other test I'll do is checking the power level in upconversion mode and comparing to the modulator requirements to check it's appropriate.

closed 2019-12-09 by hartytp