sinara-hw/Phaser #79: layout & routing review #2

gkasprow, 2019-12-09:
- [ ] move TR5A away from J17. Baluns are very fragile and will be damaged during J17 insertion. One can move J17 down.
- [ ] make R49 DNP in baseband variant
- [ ] make R257 DNP in baseband variant
- [ ] mark SW1A, SW1B TERM position (aka Sampler)
- [ ] enable designators for RN resistive networks
- [ ] mark + pin for fan connector
- [ ] just above U1 there are unknown test points. Please add legend
- [ ] separate RF traces as much as U can
[image]

- [ ] separate traces
[image]
- [ ] separate ATT_CLK from neighbour traces. It is too close to LE and RST lines and will cause issues
- [ ] the same with upconverter SPI CLK traces
[image]

gkasprow, 2019-12-09:
- [ ] the DS says that the decoupling caps should be close to the IC
[image] The example layout does not follow this advice strictly.
I would make the traces much thicker, try to pour polygons.
[image]
- [ ] make supply stubs thicker
[image]
- [ ] cosmetics
[image]

gkasprow, 2019-12-09:
- [ ] violations
[image]
- [ ] add panel to the BOM as it is done in Fastino. Use [my libraries](https://www.dropbox.com/sh/naxdly09ulgj9bt/AAAL_bULdZhdKlPEa4TrLLyba?dl=0)

gkasprow, 2019-12-09:
Please replace symbols and footprints with proper ones from CERNLIBs:
- [ ] LTC2323CUFD-16#TRPBF
- [ ] TRF372017IRGZT
- [ ] TPS72012DRVR
Make sure that no changes are done to PCB. This way we will make sure that in symbols created by me there are no errors.

gkasprow, 2019-12-09:
- [ ] the IQ channels are not perfectly matched. The GND below is OK, but above the must have GND as well. Route parts of supply planes using different layers/traces. Make sure the IQ traces have GND planes above them. Use multiple vias along the lines to couple the polygons together
[image]

gkasprow, 2019-12-09:
- [ ] I'm not sure if such mismatch makes any difference, but let's equalise them as well
[image]

hartytp, 2019-12-09:
- [ ] can we also length match between the two RF/uw outputs (not just between the I/Q of a single channel). This isn't strictly required but is always nice to do.

hartytp, 2019-12-09:
- [ ] should the ADC sampling capacitor be as close the ADC as possible to minimize high-frequency impedance?

hartytp, 2019-12-09:
- [ ] should IC18 be away from the AFE to avoid routing the noisy SMPS supply under the AFE?

hartytp, 2019-12-09:
- [ ] are these traces around the power inductors thick enough? 
[image]

gkasprow, 2019-12-09:
- [ ] some components are too close to each other. Ty to follow layer 15 as clearance marker
[image]. 
- [ ] @hartytp for these frequencies we don't care. It will sample it in diff way.
@hartytp IC18 gets clean 12V filtered by 4-th order LC filter. The net should be called N12V0A

gkasprow, 2019-12-09:
>are these traces around the power inductors thick enough?

PI analysis will tell us what current density is sufficient. In case it is too high, we will make them thicker

hartytp, 2019-12-09:
- [ ] for reliability traces should stay inside pads shouldn't they? 
[image]

hartytp, 2019-12-09:
- [ ] what's the view on vias under small components? Is that ok? 
[image]

gkasprow, 2019-12-09:
> for reliability traces should stay inside pads shouldn't they?

we have solder mask so it's not critical. 

>what's the view on vias under small components? Is that ok?

that's fine. we have soldermask and vias are covered

gkasprow, 2019-12-09:
Oh, you posted wrong picture. For QFN pads yes, the tracks should stay within pads.

hartytp, 2019-12-09:
Sorry, yes, new computer, new keyboard shortcuts...

hartytp, 2019-12-09:
- [ ] are you happy with the isolation between the MMCX clock input and the digital traces? That looks to route right through the busiest part of the board!

hartytp, 2019-12-09:
- [ ] what's the view on traces that bridge QFN pins like this? Does it impact soldering reliability? 
[image]

gkasprow, 2019-12-09:
> are you happy with the isolation between the MMCX clock input and the digital traces? That looks to route right through the busiest part of the board!

- [ ] there are only I2C lines nearby. But the clock line could be better separated from DAC LVDS lines LVDS
[image]

gkasprow, 2019-12-09:
> what's the view on traces that bridge QFN pins like this? Does it impact soldering reliability?

Where did you find them? Sure, this is a serious issue because it screws the AOI

hartytp, 2019-12-09:
- [ ] there are a few places where decoupling caps could be moved closer to the ICS to minimize inductance like here. It feels a bit odd to decouple with multiple different inductance values in parallel, but have them 5mm from the IC. That feels like a long way for a 4GHz signal
[image]

gkasprow, 2019-12-09:
- [ ] Here they are
[image]

hartytp, 2019-12-09:
> Where did you find them? Sure, this is a serious issue because it screws the AOI

Several places like IC22. Also, what's the view on traces coming out of the side of pads on QFNs? I see that in a few places (on the corner pins). It's not necessary for the layout and I didn't think it was a good idea.

gkasprow, 2019-12-09:
> there are a few places where decoupling caps could be moved closer to the ICS to minimize inductance like here. It feels a bit odd to decouple with multiple different inductance values in parallel, but have them 5mm from the IC. That feels like a long way for a 4GHz signal

this I posted already. @CannaCardo followed strictly datasheet recommendation but I think the traces should be thicker and caps placed closer

gkasprow, 2019-12-09:
it's not a problem if traces go from the QFP pad side.

hartytp, 2019-12-09:
- [ ] keep a nice big clearance between mount holes and components/traces. I've had too many cases where someone picks a bolt with a larger head than the component library assumed. 
[image]

gkasprow, 2019-12-09:
- [ ] avoid bridges between pads
[image]

hartytp, 2019-12-09:
@gkasprow what do you think about the digital/analog isolation; smps (and other nosie source) layout; keeping critical sections like loop filters compact; minimizing decoupling inductances; etc?

gkasprow, 2019-12-09:
As I wrote, I will add some slots in GND that will further improve isolation. To me, it looks that we should not have issues with them. We have 2 GND planes whith will provide low impedance for return signals.

hartytp, 2019-12-09:
@CannaCardo over all I think it's a really good layout. Good work!

A few small points above to consider, but I don't think it's far off at all. Thanks!!

hartytp, 2019-12-09:
@gkasprow great! I agree, I don't see any major issues with the layout.

@WeiDaZhang any thoughts?

CannaCardo, 2019-12-11:
> Please replace symbols and footprints with proper ones from CERNLIBs:
> 
> * [x]  LTC2323CUFD-16#TRPBF
> * [x]  TRF372017IRGZT
> * [x]  TPS72012DRVR
>   Make sure that no changes are done to PCB. This way we will make sure that in symbols created by me there are no errors.

No errors found

CannaCardo, 2019-12-11:
I've moved decoupling caps closer to IC2A / IC2B and made the traces thicker (not much space for polygons here), please confirm if that's enough. The filter components are physically large so bringing them closer would also place them near the signal path (pin 36 ext_vco)

hartytp, 2019-12-11:
> I've moved decoupling caps closer to IC2A / IC2B and made the traces thicker (not much space for polygons here), please confirm if that's enough. The filter components are physically large so bringing them closer would also place them near the signal path (pin 36 ext_vco)

Thanks! As a general point, can you have a look over the design and try to make sure that the layout is as tight as possible (minimizing parasitic inductances/loop areas etc) for all high-frequency/low noise (including things like SMPSs) areas of the board. I'll double check once the next layout RC is complete, but you've spent much more time thinking about this than me and are a better person to judge.

closed 2019-12-13 by hartytp

hartytp, 2020-02-25:
Thanks for the tours all! Excited to see this spooled 
[image]

hartytp, 2020-02-27:
Did this come off the production line yet?

gkasprow, 2020-02-28:
Yes, today.

hartytp, 2020-02-28:
Cool! Photo? :)