sinara-hw/Phaser #81: polygon crossings

jordens, 2019-12-14:
RF BB

[image]

GT pairs

[image]

jordens, 2019-12-14:
I would have just put all the high frequency signals (dac differential, dac baseband to switch, GT links) that don't fit on the top layer on the bottom layer and not some inner layer. that way you only need to look at one adjacent copper plane and not two. and the via stubs are also gone (but this is probably not that critical at ~4 GHz yet).

gkasprow, 2019-12-15:
I managed to cover all DAC analog lines with GND on both sides.
I don't care that much about GTH since they are coupled mostly with the layer5. I optimized the planes to minimize Z discontinuity.

gkasprow, 2019-12-15:
We would need to add at least one more layer to fully support GTP lines. But since we won't probably use them at all, it's not worth the effort. Moving them to the bottom layer is not that easy

hartytp, 2019-12-15:
Great

closed 2019-12-15 by gkasprow

jordens, 2019-12-15:
Sounds good. We should just do a loopback sata and check the SI with the Bert scanner.