sinara-hw/Phaser #83: upconverter refin transformers

jordens, 2019-12-14:
i still don't understand why these need to be transmission line transformers and can't be of the 1:4 (dac signal path) or 1:2 (clock inputs) type we use elsewhere.

also, maybe populate the 1:2 baluns on the clock inputs by default (and isolate). @hartytp

hartytp, 2019-12-14:
> i still don't understand why these (TR5A TR5B) need to be transmission line transformers and can't be of the 1:4 (dac signal path TR3 TR8 TR4 TR5) or 1:2 (clock inputs TR1 TR2) type we use elsewhere.

Agreed, it would make more sense to use the same baluns used elsewhere in the clock network. Using DC-isolating transformers and corresponding layout would also be nice.

One comment however: TC2-1TX+ max input frequency is only 300MHz. That's a relatively low BW since we want to maintain >1V/ns rising edge, but the clock input may be relatively low amplitude after cable losses etc. Is there a slightly faster balun we can use for the clocking? I.e be happy accepting a slightly higher min frequency as a compromise, but if we could get to closer to 1GHz BW I'd be happy.

hartytp, 2019-12-14:
With 300MHz we'd ideally need a ~1Vpp clock input to get a 1V/ns edge time (very crude estimate based on `2*pi*A*f`-type maths), which is more than Kasli ref puts out IIRC...

jordens, 2019-12-14:
It's 0.3 dB at 300 Mhz worse than at 150 MHz. The 3 dB bandwidth which you'd use for slew rate estimates is higher.

hartytp, 2019-12-15:
Yes, but I suspect the performance goes off a cliff not much above 300MHz. Anyway it’s probably be fine but is there any reason not to replace all clock baluns with TC2-72T+

jordens, 2019-12-15:
Very likely not a cliff. More like 3 dB at 600 MHz would be my guess without looking at the s params.
The TC2-72T+ is also fine. More loss throughout the band but wider high frequency behavior.

hartytp, 2019-12-15:
I agree either is prob fine. The 72T+ would be my preference but it’s not a big deal either way

jordens, 2019-12-16:
Ok. TC2-72T+ for TR5A, TR5B, TR1 and TR2 then, all four populated by default on both variants?

hartytp, 2019-12-16:
Works for me! Thanks for raising this @jordens

gkasprow, 2019-12-16:
fixed

closed 2019-12-16 by gkasprow