sinara-hw/Phaser #86: u.fl and SMT MMCX

jordens, 2019-12-14:
* Are you confident the SMT MMCX are mechanically robust enough? IME SMT connectors are a reliability issue (see the typical micro-USB failures).
* If you are happy with the SMT MMCX, then I'd really use them for the debug connectors as well (J4A J4B)

gkasprow, 2019-12-16:
they need vias that attach pads to inner layers and polygon above to be robust.

jordens, 2019-12-16:
If that works I'm happy with it.

 I thought you needed a pin or wing through the pcb together with it's solder to protect from pad shear.

jordens, 2019-12-16:
On the other hand that pretty much kills the notion of SMT connectors not piercing through all layers. The vias may actually make the effective footprint of a SMT connector larger than the through-hole style.

jordens, 2019-12-16:
I'll leave the u.fl and MMCX connector style choice to you then.

gkasprow, 2019-12-16:
I changed ufl to MMCX to reduce the BOM

closed 2019-12-16 by gkasprow