sinara-hw/Phaser #93: keepout clearance

jordens, 2019-12-14:
Just confirm that you are happy with these collisions between the clearances and between clearance and keeout:

[image]

[image]

[image]

[image]

[image]

gkasprow, 2019-12-15:
some of them are top and bottom overlapping. Some need fixing.

gkasprow, 2019-12-15:
Some are fine, some I fixed.

closed 2019-12-15 by gkasprow