sinara-hw/Phaser #94: PI analysis

gkasprow, 2019-12-15:
Recently I purchased a commercial license for CST PDN analyzer. It is an Altium plugin.
2 years ago it was a simple tool, but with recent release, it copes well with SMPS, LDOs,  chokes, etc. One can simulate the entire power domain. Moreover, one can set individual tolerances for each of ICs. 
The tool is much cheaper than Hyperlynx PI (5k instead of 60k). It is tightly integrated with ADI so one doesn't have to export and then import the hyp file. It does not have 3D mode but instead has a lot of result evaluation tools that fully compensate it. and result browsing and bug fixing is much faster.
A few screenshots
[image]
[image]

hartytp, 2019-12-15:
Nice! Good to know about that tool—I always do PI-by-eye so it’s good to know of an affordable professional tool to do the job properly!

gkasprow, 2019-12-15:
PI has shown that IC29 has too low current. I replaced it with 1A one TPS79625DCQG4

gkasprow, 2019-12-15:
[image]

gkasprow, 2019-12-15:
IC21 has too low Imax. In the power budget I included only single upconversion chip.

hartytp, 2019-12-15:
Good catches

gkasprow, 2019-12-15:
The DS says that max power consumption is 148mA. But when one adds max power ratings from all rails we get much more than that. In the power budget, I used this lower value. Let's use these max ratings to make sure we won't get issues.

gkasprow, 2019-12-15:
I will use [TPS78650](http://www.ti.com/lit/ds/slvs389m/slvs389m.pdf)

gkasprow, 2019-12-15:
It's easier than adding second LT3045 and output matching resistors

gkasprow, 2019-12-15:
[image]

gkasprow, 2019-12-15:
I had to move the 1V SMPS feedback connection. Otherwise, the drop will cause issues. It's impossible to place solid polygon with 6 layers.

gkasprow, 2019-12-15:
The SMPS simply senses the voltage under the FPGA.

gkasprow, 2019-12-16:
The PI looks fine. The only place the current density is exceeded is the VCCINT plane under the FPGA running at 3A. The voltage is however fine. But I don't expect any issues. It's not worth adding more layers.

hartytp, 2019-12-16:
Cool! Shall we close this?

closed 2019-12-16 by gkasprow

hartytp, 2019-12-17:
I assume you also checked things like inductor current ratings etc as part of this? I haven't explicitly checked that during my review.

gkasprow, 2019-12-17:
Yes, I entered a detailed PDN model including LDO max current, inductor currents, SMPS efficiencies, etc.
It takes a few hours but then the tool checks all at once in a few minutes including drops, I density, via current, etc.
For this reason, I had to replace some ferrite beads because I exceeded the current limit or voltage drop.
The same applies to some LDOs.
This tool is really cool, much more advanced than Hyperlynx, at least in DC drop calculation.
It is not that stable and sometimes generates errors. I opened one service request.