sinara-hw/Phaser #95: review new SMPSs

hartytp, 2019-12-16:
A few changes were made to the SMPSs during PI analysis. I want to do a final check on the design before it goes to manufacture to double check for any mistakes (e.g. the converter only has Vin max of 6V so need to check the input voltage etc).

hartytp, 2019-12-17:
Checked IC29 and IC21 per https://github.com/sinara-hw/Phaser/issues/94 and didn't spot any issues.

@jordens assuming you're happy with the design as well, so we can go to manufacture.

Thanks so much @gkasprow @jordens for all the work on this! Really looking forward to getting my hands on HW in the New Year.

@gkasprow can you remind me what the lead time should be for this?

closed 2019-12-17 by hartytp

jordens, 2019-12-17:
There are a couple uncovered vias. Two of them at the clock input from the cpcis adapter may or may not suck the solder from the capacitors. Other than that this looks really good. Excellent copper layer management.

hartytp, 2019-12-17:
And of course a massive thanks to @CannaCardo for the lovely layout!

gkasprow, 2019-12-17:
@about the lead time ask Technosystem