sinara-hw/Phaser #96: IC21 Ven

hartytp, 2019-12-17:
@gkasprow actually, I did spot one thing!

IC21 Ven is driven from 1V5. However the data sheet requires a min of 1.7V

[image]

hartytp, 2019-12-17:
TPS74901RGWT is okay with 1V1 so the other supplies on this rail are fine.

[image]

hartytp, 2019-12-17:
@gkasprow other than that I don't see any issues with the two regulator changes, but let me know if there is anything else I should look at.

gkasprow, 2019-12-17:
OK, so I have to add a pullup resistor.

gkasprow, 2019-12-17:
[image]

closed 2019-12-17 by hartytp