sinara-hw/Phaser #101: v1.0 tests

gkasprow, 2020-03-04:
- after  correction of #100 the power supply values are OK
- the power rail ripples are below 10mV on digital rails and at the scope noise level on analog rails.
- the power consumption for baseband version without FPGA bitstream is 2.4W
- the power consumption for the upconverting version without the FPGA bitstream is 3.8W

gkasprow, 2020-03-04:
the FPGA is correctly detected over JTAG
`INFO: [Labtools 27-1435] Device xc7a100t (JTAG device index = 0) is not programmed (DONE status = 0).`

hartytp, 2020-03-04:
Awesome! Will you be able to get rf out of the dac using Xilinx ip?

gkasprow, 2020-03-04:
That's a bit tricky. DDS is trivial with Xilinx IP tools, but DAC initialization seems to be much more complex (48 16-bit registers)

hartytp, 2020-03-04:
Ok. @jordens anything else you want done before shipping?

gkasprow, 2020-03-04:
however, I can test the upconverter because I have a devkit and can connect the SPI lines directly to the mixer chip. But I need to get something on the DAC outputs first

hartytp, 2020-03-04:
> That's a bit tricky. DDS is trivial with Xilinx IP tools, but DAC initialization seems to be much more complex (48 16-bit registers)

I wonder how many actually need to be configured beyond their reset values to get basic functionality.

gkasprow, 2020-03-04:
the eval software seems to do the config job

gkasprow, 2020-03-04:
the eval SW works only with the connected devkit. I did not find a way how to generate the register map.

gkasprow, 2020-03-04:
ok, that's not that bad. TI delivers several default config files. This is example one
[DAC34H84_FDAC_1228p8MHz_4xint_NCO_30MHz_QMCon.txt](https://github.com/sinara-hw/Phaser/files/4289001/DAC34H84_FDAC_1228p8MHz_4xint_NCO_30MHz_QMCon.txt)

hartytp, 2020-03-04:
Nice!

gkasprow, 2020-03-04:
I plan to play with them. The idea is to use Xilinx DDS Compiler that outputs sin/cos 16-bit value, then use DDR primitive to realize the output interface. 100MHz would be sufficient to do some tests without waiting too long for compilation.
Then use FORTH embedded CPU that does all DAC SPI configuration. To config the upconverters, I will route their SPI lines to EEM and then use tester EEM to connect to the TI devkit where I get nice GUI to access to all registers
 I can build such a setup quickly.

hartytp, 2020-03-05:
Perfect!

jordens, 2020-03-05:
Checking that the LVDS interface works at full required speed would be helpful (using the pattern checker). Debugging that is not something I'm set up for.
AFAICT there is no need to generate signals on FPGA for testing the RF chain. You can just output constant and use the DAC NCO to upconvert that.

gkasprow, 2020-03-05:
True, I'll have a look at it.

gkasprow, 2020-03-05:
We need the production PTS anyway, so It's good to make the pattern checker running anyway.

jordens, 2020-03-05:
It'll likely also be part of the standard init sequence in ARTIQ and our test suite as well. Probably worth doing it in such a way that there is maximal code reuse.

gkasprow, 2020-03-05:
Yes, we do it with ARTIQ. The LEDs are blinking :)

jordens, 2020-03-05:
As you know, we have also had a test suite for a long time and every bit of possible self-testing functionality that can be moved from private one-shot production and integration tests into ARTIQ proper is extremely useful for everybody including manufacturing. It increases coverage of the tests, of the hardware, and allows the test suite to be maintained together with ARTIQ preventing divergence.

gkasprow, 2020-03-09:
@hartytp @jordens we are working on basic firmware support that will let us characterize the design.
TS wants to send all 6 boards to you immediately. What is the decision then? I'd need 2 boards to do initial tests. I don't want you to do HW debugging.

jordens, 2020-03-09:
No point in shipping them to me right now. I can't do much with the boards yet anyway. First need to make progress on the gateware development for it. That will be another few weeks. Since shipping is fast, I'd prefer you keep the boards for the next two weeks. Then we revisit.

hartytp, 2020-03-09:
I'm happy for TS to send me an invoice now if easier? I agree with @jordens though, no point shipping before basic testing is complete.

hartytp, 2020-03-21:
@gkasprow any progress with this or is it blocked by the uni being closed?

gkasprow, 2020-03-21:
Mikolaj is working on it all the time. We have access to the WUT but we go there only when necessary.

hartytp, 2020-03-21:
Wonderful! Let me know when you have results

kaolpr, 2020-03-21:
Hi, indeed I work on testing of the Phaser. Some remote-working and epidemic related difficulties stopped me for a while, but I can confirm that basic gw for Phaser FPGA is ready and is being tested. Hoping to have it tested at the end of the next week at the latest.

hartytp, 2020-03-21:
Awesome!

kaolpr, 2020-03-30:
Well, referring to my previous post, regrettably I have to report that tests are not yet finished. Currently all chip communication works and I'm experimenting with DAC LVDS interface checker. Hope to have some results in the days to come.

hartytp, 2020-03-30:
Glad to hear things are progressing!

hartytp, 2020-04-06:
@kaolpr how's the testing going? We're really looking forward to getting our hands on Phaser!

kaolpr, 2020-04-06:
Fully understand. However, I'm still having some difficulties with DAC and need to do some on site investigation (what is not that obvious nowadays). I'll do my best to give ship hardware to you at most at the beginning of the next week. I know that you heard that already, but circumstances does not make things easier.

hartytp, 2020-04-06:
Don't worry, we're all in the same boat!

HW at the beginning of next week would be ace.

kaolpr, 2020-04-10:
Ok, so DAC LVDS interface is working now. During weekend I'll try to get RF output and hoping to have board ready at the beginning of the week (rather Tuesday, as Monday is Easter Monday).

hartytp, 2020-04-10:
Awesome!

kaolpr, 2020-04-15:
Heh... I need to go to the lab once more and play with oscilloscope in the signal path as I can't get RF output. I'll do my best to do it tomorrow.

kaolpr, 2020-04-16:
DAC is working, but still can't get TRF3720 working (no lock, even on devkit). Will try again tomorrow.

hartytp, 2020-04-17:
> DAC is working

Great!

>  still can't get TRF3720 working (no lock, even on devkit). Will try again tomorrow.

Thanks for the good work! Getting there...

hartytp, 2020-04-20:
@kaolpr can you post any data on the performance without the DAC (e.g. a spectrum for a single tone on the DAC) just to show basic operation.

Once that's done, please can you ship a baseband board to @jordens asap to begin gateware/firmware development?

The upcovnersion variant is less crucial since we need the baseband firmware first.

gkasprow, 2020-04-21:
I have the boards. Since I'm heavily overloaded with another big project (COVID related), I will post the boards to @jordens tomorrow. I will keep one board to work with upconverter.

kaolpr, 2020-04-29:
I've published my codes for Phaser FPGA: https://github.com/elhep/phaser_gw and corresponding ARTIQ support: https://github.com/elhep/artiq/tree/elhep_phaser

hartytp, 2020-04-29:
@jordens did you receive a phaser from Greg yet?

jordens, 2020-04-29:
No

hartytp, 2020-04-29:
@gkasprow any idea where this is? Did you send it? Do you have a tracking number?

gkasprow, 2020-04-29:
I handled it to TS.

jordens, 2020-04-30:
@kaolpr Do you get RF output from that gateware? Could you post spectra?
Also could you run that testbench on all the prototype boards that are about to be sent out?

kaolpr, 2020-04-30:
Nope. I faced a problem with configuration of modulator. Please do not treat these codes as a fully working. I checked that I was able to get some output from the DAC and that LVDS interface tests pass.

gkasprow, 2020-04-30:
I have a devkit for the modulator chip. I can check something quickly, but at the moment have other very urgent project.

jordens, 2020-04-30:
@hartytp do you want to risk it? I won't be able to do much hw debugging.

hartytp, 2020-04-30:
> @hartytp do you want to risk it? I won't be able to do much hw debugging.

To check we're all on the same page: I'm not worried about the modulator for now; just the baseband functionality.

> I checked that I was able to get some output from the DAC and that LVDS interface tests pass.

What output did you get out of the DAC? Was it an RF waveform? Glad to hear that the LVDS interface tests passed, but it would be good to know what tests you did on the output. I assume you at least checked that you could get something that looked like a sine on a scope out of the SMAs?

hartytp, 2020-04-30:
So long as we have seen a basically sinusoidal output I'm happy proceeding with firmware development. Once we have the boards we can flash with your test code and measure spectra etc to check the performance in slightly more detail...

kaolpr, 2020-05-01:
@hartytp I only had access to the version with modulator at that time. I got a sine output that I looked in between DAC and modulator with a scope. Unfortunately I did not managed to get modulator working (but I think that must be a software issue).

hartytp, 2020-05-01:
> I got a sine output that I looked in between DAC and modulator with a scope

Okay, that's a good start! So, we do have by-eye-good RF out of the DAC. What frequency/

> @hartytp I only had access to the version with modulator at that time

If TS give you a baseband board, could you repeat the test while looking at the SMA output RF at, say, 100MHz on a spectrum analyzer?

If not, I'll see if someone here can test that quickly (@dnadlinger ?)

dnadlinger, 2020-05-01:
> If not, I'll see if someone here can test that quickly (@dnadlinger ?)

I've got access to Kasli and a scope, so a few quick, basic checks I could do (if I had hardware). I wonder how much that would help, though, as I wouldn't be able to debug any actual hardware issues either.

hartytp, 2020-05-01:
Thanks!

@kaolpr if you can get a spectrum out of the baseband unit it some point next week that would be really appreciated!

hartytp, 2020-05-05:
Ping @kaolpr

kaolpr, 2020-05-05:
I'll try to do it tomorrow, but I'm not able to reach lab this week, so if
something goes wrong remotely I won't be able to debug.

kaolpr, 2020-05-08:
[image]
That's what I got. I guess there may be some digital issue with signal generation. Green waveform is output at SMA, purple is MMCX DAC output.
I've also updated gateware repo as I realized that support for attenuator control was missing.

hartytp, 2020-05-09:
What's the DAC clock, RF frequency etc?

gkasprow, 2020-05-09:
I'd run it wich much lower frequency. then the scope /probe BW won't distort the original artifacts.

kaolpr, 2020-05-10:
@hartytp DAC clock was 125 MHz (from Kasli), data generator was also working with 125 MHz and data patter length was 128, so expected output frequency was ~976 kHz. Samples were half of the FS range.

@gkasprow I guess it was well within the scope BW.

hartytp, 2020-05-10:
@gkasprow @jordens the scope traces don’t look how I’d expect. What do you want to do next?

[44 more replies]

closed 2020-08-28 by jordens