sinara-hw/Phaser #106: SMT mmcx connectors

jordens, 2020-08-27:
I managed to break off the clock input SMT MMCX connectors (as feared in #86) on the first try of removing a connector. That was one of the low insertion/extraction force connectors. While the pads (apart from the center one) were held by the vias, the solder joints broke.
Can we please at least make the clock MMCX connector a through-hole one?

gkasprow, 2020-08-27:
sure, does it apply to the other designs?

jordens, 2020-08-27:
No. The others all have through hole mmcx.
I'm also fine ufl for the connectors around the upconverters. They are more experimental and making them through hole may complicate routing.

gkasprow, 2021-03-11:
we can also glue them, it's an automatic process.

gkasprow, 2021-03-11:
changing J13, J14, J15, J17 to u.fl
changing J7 to THT one

closed 2021-03-11 by gkasprow