sinara-hw/Phaser #107: fan pwm

jordens, 2020-08-28:
I can't seem to bring the gate low enough to turn the fan off. This is with a 1 (or 0.25) MHz PWM and any duty cycle.
@gkasprow how is this supposed to work?

gkasprow, 2020-08-28:
the idea is that you cannot turn it off with logic level. You must use PWM. 
However, the BSH103 has a very low Vth. Originally BSS138 was used, then it was replaced probably to unify the BOM. 
If the circuit does not work with very low duty cycles, it may be caused by very low Vth of this particular T20. You can try to replace it with BSS138.
The circuit does just a DC-restoration, that makes sure the logic pattern on the FPGA appears on T20 unchanged, only shifted up by roughly 0.2V (Schottky diode Vf). When FPGA is not toggling, after the RC time constant, T20 starts conducting.
You can also short C159 if you don't care about such protection.

gkasprow, 2020-08-28:
You can also replace T20 with BSS138

jordens, 2020-08-28:
That's what I thought. The minimum gate voltage is above 0.7 V already at 0.5 duty cycle and increases below that.
Then we'll either have to change the FET or RC to make it work.

gkasprow, 2021-03-11:
changed to BSS138

closed 2021-03-11 by gkasprow