sinara-hw/Phaser #109: power budget: OPA2197

hartytp, 2020-08-28:
there are two sets of OPA2197. Both are listed as being supplied from +-11V5 in the power budget, which is not correct

[image]
[image]
[image]

gkasprow, 2020-09-24:
good catch, fixed

dnadlinger, 2020-09-25:
As a minor detail, also seems like those feedback resistors are unnecessary, or at least unnecessarily large, on the reference buffers.

gkasprow, 2021-03-11:
they match the resistance visible by the positive opamp input

closed 2021-03-11 by gkasprow

dnadlinger, 2021-03-11:
But the OPA197 bias current is 20 pA max., so they only serve to increase voltage noise for little benefit?

gkasprow, 2021-03-11:
in this case, you are right.