sinara-hw/Phaser #111: RAM memory test

hartytp, 2020-09-16:
@gkasprow during your initial testing, did you run a mem test/SI test on the RAM on phaser? It's worth checking this before the next hardware revision.

gkasprow, 2020-09-16:
no, we didn't do that AFAIR. This can be easily done with Xilinx MIG. It's the same SDRAM as in Kasli, but anyway, it's worth checking with Xilinx tool because it provides in depth diagnostics, eye diagrams etc

gkasprow, 2020-09-25:
the SDRAM works at 400MHz. I'm running the XIlinx test generator with MIG, and cannot trigger it on dgb_cmp_error signal which means this signal does not toggle.
[image]

gkasprow, 2020-09-25:
[image]

hartytp, 2020-09-25:
goood!

closed 2021-02-25 by hartytp