sinara-hw/Phaser #112: tune loop filters

hartytp, 2020-09-16:
we should check the loop filters for the upconver and DAC before moving to the new release (e.g. to make sure that there isn't massive ringing etc).

jordens, 2020-09-16:
Just the upconverters.

hartytp, 2020-09-16:
why not the DAC? Have you checked that already and are happy with it?

jordens, 2020-09-16:
There is no loop filter that could be checked.

hartytp, 2020-09-16:
aah, okay, I'd forgotten that we're using the integrated loop filter.. Anyway, we should still check the close-in spectrum to check for servo bumps etc.

hartytp, 2021-03-02:
We do plan to do this once we receive hardware with an upconverter. However, since it's only a BOM change, I'm happy sorting this out after the v1.1 boards have gone to print.

gkasprow, 2021-03-11:
any conclusion here?

hartytp, 2021-03-12:
haven't been able to check yet, but it shouldn't be a release blocker. Let's leave it open for now

kaolpr, 2026-09-17:
Assume that if there was no comments in last 5+ years the problem is sorted out/non existent. Feel free to reopen if you have any new information.

closed 2026-09-17 by kaolpr