sinara-hw/Phaser #114: check IQ performance

hartytp, 2020-09-16:
verify that carrier nulling, sideband suppression etc are all roughly within spec

hartytp, 2021-03-02:
Personally, I'm happy leaving the bigger FPGA since IIRC it doesn't make a huge difference to the overall cost, and gives us more space to experiment with different designs in the future. However, I'm happy to leave the final call to @jordens if he has stronger opinions.

gkasprow, 2021-03-11:
shall I close it?

jordens, 2021-03-12:
Leave the current one. The stft needs a bit more resources.

hartytp, 2021-03-12:
oops, I posted in the wrong thread.

Re FPGA size: agreed, let's stick with the one we have.

Re checking IQ performance: let's leave this open until someone has checked it, but it shouldn't be a release blocker (right now there isn't a clear pass/fail specification for this anyway)

pathfinder49, 2021-04-09:
I've measured the uncompensated Carrier feedthrough (LO leakage) and sideband suppression with the LO at 2.875 GHz. The uncompensated sideband suppression was -38 dBc. The carrier feed-through was -37.5 dBm. Nominally, these are close to typical values. The regarding the feed through, I'd like to understand #130 before deciding this is to spec.
[image]

pathfinder49, 2021-05-14:
This performance is similar after addressing the LO output amplitude. Playing with the DAC QMC settings I saw siedband powers of -45 dBc. Adjusting the TRF `ioff` and `qoff`, I could reduce the LO loakage to -68 dBm.

kaolpr, 2026-09-17:
Closing due to inactivity. @pathfinder49 @hartytp If you have any new data/conclusions feel free to reopen.

closed 2026-09-17 by kaolpr