sinara-hw/Phaser #116: check power consumption

hartytp, 2020-09-16:
seems a bit high on some rails indicating a potential problem: https://github.com/sinara-hw/Phaser/issues/101#issuecomment-682044877

gkasprow, 2020-09-24:
I measured power consumption, but since there is some issue with DAC, they may not be representative for all rails.
P12V0A 3.2mV 0.24R I= 13mA
N12V0A 4.8mV 0.24R I=20mA
P5V5 6.4mV 0.045R I= 142mA
P3V6A 13.7mV 0.045R I=304mA
P1V5A 6mV 0.045R I=133mA
P3V3 30mV 0.045R I= 666mA
P1V0 14mV 0.045R I= 311mA

gkasprow, 2020-09-24:
Not symmetrical current consumption of the N12V0A is caused mainly by the LED, which has 1.2k resistor.

gkasprow, 2020-09-24:
I used inductor resistance to estimate the currents, so there is something like a 10% error.

gkasprow, 2020-09-24:
I see one issue with 3V3 rail consumption. I assumed max value of 384mA, but I measure more than 660mA

gkasprow, 2020-09-24:
P1V8A consumes I= 4.8mV/0.05R = 96mA which is fine

hartytp, 2020-09-24:
@gkasprow was this for the baseband or upconverting module?

> P5V5 6.4mV 0.045R I= 142mA

I don't understand the power budget. What are "from dep." and "with dep."? How does this current compare to expectations?

hartytp, 2020-09-24:
@gkasprow any idea why @jordens measured the  TRN 3-1222 getting to 65C when it's not dissipating much power at all?

gkasprow, 2020-09-24:
P3V3 without FPGA config is just 9.3mV/0.034=206mA. 
With FPGA configured, it consumes 666mA. This rail supplies 2V5 and 3V3 FPGA rails.

gkasprow, 2020-09-24:
@hartytp "from dep" means it is supply from dependent rails, i.e. LDOs.
"with dep" is a sum of LDOs and main rails.

gkasprow, 2020-09-24:
the TRN 3-1222 consumes 40mA of idle current. Which is already half of watt. It is well isolated thermally from the PCB so it's not suprising that it heats itself to such temperature.
The load is 165mW which adds roughly 30..40mW to the bill.

gkasprow, 2020-09-24:
we have 37 LVDS signals. If we assume each of them is 2x4mA, we get 296mA of just static power, just for LVDS links

gkasprow, 2020-09-24:
that I didn't take into account making the power budget

gkasprow, 2020-09-24:
but it does not change the power budget so much 
[image]

hartytp, 2020-09-24:
@gkasprow with the caveat that the DAC isn't fully configured, what's your conclusion? Does the measured total/rail power match the budget?

gkasprow, 2020-09-24:
For the moment I have
4.7W - Kasli
9.2W Kasli + Phaser
12.5W Kasli + Phaser with FPGA configured
12.8W after running Phaser_playground.py
I would be able to check more when I have the DAC running.

jordens, 2020-09-24:
125 mhz clock from kasli connected?
And is that face alarm happening always or just the first time you run the script?

gkasprow, 2020-09-24:
Yes, it is. I see it on the scope.
The alarm error appears each time I run the script.
However, this is the board that didn't want to work with testbench @kaolpr wrote. I expect it has some HW issue with FPGA-DAC.

jordens, 2020-09-25:
Maybe. But you are also the first one to test it afaik. So there are likely issues to be found. Could you print out the dac alarm register in `check_dac_alarns()` before the error is raised?

gkasprow, 2020-09-25:
sure. How exactly should I do it? I'm a newbie to ARTIQ

jordens, 2020-09-25:
I added [a commit](https://github.com/m-labs/artiq/commit/2fba3cfc78252d2b87d4e005099deee4d08c60dc) to make this a bit easier (no need to make new bitstreams but double check that artiq phaser tree is the latest and that you are using the phaser v0.3 bitstream):

* make the call to Phaser [`.init()`](https://github.com/quartiq/phaser/blob/9b9e0de2b29523453f9c60ec14fbd1e64273e23c/example.py#L30) in your script a `.init(debug=True)` and it will just print the alarm register and continue regardless
* Run your script twice and post the number it prints here.
* If the [alarm](https://www.ti.com/lit/ds/slas751d/slas751d.pdf#page=65) is one of the three fifo collision errors, play with the `fifo_offset` in your device database `device_db.py` (values 0-7), like:

```python
...
device_db["phaser0"] = {
    "type": "local",
    "module": "artiq.coredevice.phaser",
    "class": "Phaser",
    "arguments": {
        "channel_base": ...,
        "dac": {"fifo_offset": 2},
        "clk_sel": 0
    }
}
```

* There should be two or three values of `fifo_offset` that don't lead to fifo errors.

gkasprow, 2020-09-25:
`
[nix-shell:/workspace]$ artiq_run ./experiments/phaser_playground.py
2112
Core Device Traceback (most recent call last):
  File "./experiments/phaser_playground.py", line 12, in __modinit__ (RA=+0x8c8)
    self.phaser0.init(debug=True)
  File "<artiq>/coredevice/phaser.py", line -1, in ... artiq.coredevice.phaser.Phaser.init<artiq.coredevice.phaser.Phaser>(...) (RA=+0x256c)
    <unknown>
  File "<artiq>/coredevice/phaser.py", line 283, column 17, in ... artiq.coredevice.phaser.Phaser.init<artiq.coredevice.phaser.Phaser>(...)
    raise ValueError("DUC+oscillator phase/amplitude test failed")
    ^
ValueError(0): DUC+oscillator phase/amplitude test failed
Traceback (most recent call last):
  File "/nix/store/ks8m3p0isn9l87795ap258l4abc5p9g9-python3.7-artiq-6.7267.4340a5cf.beta/bin/.artiq_run-wrapped", line 9, in <module>
    sys.exit(main())
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 225, in main
    return run(with_file=True)
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 211, in run
    raise exn
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 204, in run
    exp_inst.run()
  File "/modules/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/modules/artiq/artiq/coredevice/core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 629, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 621, in _serve_exception
    raise python_exn
ValueError: DUC+oscillator phase/amplitude test failed
`

gkasprow, 2020-09-25:
each time I get the same error

gkasprow, 2020-09-25:
when I comment this part:

            # allow ripple
           # if (data_i < sqrt2 - 30 or data_i > sqrt2 or
            #        abs(data_i - data_q) > 2):
             #   print(ch)
             #   raise ValueError("DUC+oscillator phase/amplitude test failed")

it starts blinking LEDs and power consumption is 15W with Kasli

gkasprow, 2020-09-25:
It prints channel 0

gkasprow, 2020-09-25:
How to get something at the DAC output?
I tried this sequence;
        self.core.reset()
        self.phaser0.init(debug=True)
        self.phaser0.set_cfg()
        self.phaser0.channel[0].set_nco_frequency(10000000)

jordens, 2020-09-26:
Use the phaser example.py script (linked above) or the artiq_sinara_tester.py tool (in artiq/frontend)

jordens, 2020-09-26:
That phase amplitude test should not fail if you use recent bitstreams. And apparently now your dac alarms are gone.

jordens, 2020-09-26:
The phase amplitude test also fails if you hit https://github.com/quartiq/phaser/issues/5. The current work around is to reload the Phaser bitstream (or power cycle kasli+phaser if both are flashed with the desired bistreams).

gkasprow, 2020-09-26:
This time I downloaded the bitstream directly from https://github.com/quartiq/phaser/releases/tag/v0.3

jordens, 2020-09-26:
The DAC alarm you are seeing is `alarm_fifo_2away`. Try `"fifo_offset": 0` or `4` (instead of the default `2`) in `device_db.py` (see above).

gkasprow, 2020-09-27:
OK, I'm playing with example.py
When I set fifo_offset to 3, the DAC error disppears.
I get

```
./phaser/example.py:30:9-30:15: error: this function of type method(fn=(self:<instance artiq.coredevice.phaser.Phaser>, ?debug:'a)->'b delay('c), self=<instance artiq.coredevice.phaser.Phaser>) does not accept argument 'clk_sel'
        f.init(clk_sel=1)
        ^^^^^^           
./phaser/example.py:30:16-30:25: note: extraneous argument
        f.init(clk_sel=1)
               ^^^^^^^^^ 
```

jordens, 2020-09-27:
That's an old code version.

gkasprow, 2020-09-27:
OK, true.
This time I get:

```
[nix-shell:/workspace]$ artiq_run ./phaser/example.py 
['0x350068']
['0xffffffa9']
['0xffffff8a']
['0xb']
['0x4b00300c']
['0xffffff8d']
['0xff90100e']
['0xd041100f']
['0x340068']
['0xffffffa9']
['0xffffff8a']
['0xb']
['0x4b00300c']
['0xffffff8d']
['0xff90100e']
['0xd041100f']
['0x40']
Traceback (most recent call last):
  File "/nix/store/ks8m3p0isn9l87795ap258l4abc5p9g9-python3.7-artiq-6.7267.4340a5cf.beta/bin/.artiq_run-wrapped", line 9, in <module>
    sys.exit(main())
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 225, in main
    return run(with_file=True)
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 211, in run
    raise exn
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 204, in run
    exp_inst.run()
  File "./phaser/example.py", line 16, in run
    self.do()
  File "/modules/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/modules/artiq/artiq/coredevice/core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 627, in serve
    self._serve_rpc(embedding_map)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 527, in _serve_rpc
    args, kwargs = self._receive_rpc_args(embedding_map)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 385, in _receive_rpc_args
    value = self._receive_rpc_value(embedding_map)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 378, in _receive_rpc_value
    return receivers.get(tag)(self, embedding_map)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 86, in _receive_list
    item = fn(kernel, embedding_map)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 148, in <lambda>
    embedding_map.retrieve_object(kernel._read_int32()),
  File "/modules/artiq/artiq/compiler/embedding.py", line 118, in retrieve_object
    return self.object_forward_map[obj_key]
KeyError: 2895
```

jordens, 2020-09-27:
That's an old ARTIQ version (mismatch between bitstream and host code). But the error is late enough that you should have output from the DAC and the TRFs.

gkasprow, 2020-09-27:
I don't see any output. Which commits of artiq and phasern shoul I use? For the moment I'm using:
artiq:
  2fba3cfc78252d2b87d4e005099deee4d08c60dc
  d517d5d4421ae9f31ae556f458fdb2f63e4d1abd
  f0959fb8
  master
* phaser

phaser:
  2fba3cfc78252d2b87d4e005099deee4d08c60dc
  cd2a893
  d517d5d
* d517d5d4421ae9f31ae556f458fdb2f63e4d1abd
  master

jordens, 2020-09-28:
If the asterisk marks your HEAD, then the Phaser commit is ok (use the bitstream and example from that). It doesn't tell me what ARTIQ commit you are using. In any case you need to make sure that the bitstreams on the devices also match those commits, not just the code on your computer.
I've pushed v0.4 with explicit ARTIQ and misoc commits that I tested.

jordens, 2020-09-28:
Also why does the host-gateware-firmware mismatch detection seem to be [broken](https://github.com/sinara-hw/Phaser/issues/116#issuecomment-699660588)? @sbourdeauducq ?

jordens, 2020-09-28:
If you run the script to completion, comment out [this](https://github.com/quartiq/phaser/blob/d517d5d4421ae9f31ae556f458fdb2f63e4d1abd/example.py#L63) line to not put the DAC and the TRFs to sleep.

sbourdeauducq, 2020-09-28:
Works for me...

```
sb@vulcan ~/a/a/e/kc705_nist_clock (master)> artiq_run repository/mandelbrot.py
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (6.7372.a6523995.beta) and software (6.7370.29c940f4.beta) versions
               ...............................................................
             .....................,,,,,,,,,,,.................................
            .............,,,,,,,,,,,,,,,,,,,,,,,,,,,,,........................
          ..........,,,,,,,,,,,,,,,,,,,,,,,----------,,,,,,...................
         ........,,,,,,,,,,,,,,,,,,,,,------::;h+;::-----,,,,,................
       .......,,,,,,,,,,,,,,,,,,,,,--------:::;+MMhM :------,,,,,.............
[...]
```
Do you have a repro?

kaolpr, 2020-09-30:
@gkasprow asked me to help him with this. We've managed to get rid of this `KeyError: 2895` error (wrong MiSoC version). Now example script from phaser gw repo finishes with no errors. However, Greg reports no output on DAC.

I tried to use test mode by doing:
```
f.channel[0].set_duc_cfg(select=1)
f.channel[1].set_duc_cfg(select=1)
while True:
    f.channel[0].set_dac_test(0x0)
    f.channel[1].set_dac_test(0x0)
    delay(500*us)
    f.channel[0].set_dac_test(0xFFFFFFFF)
    f.channel[1].set_dac_test(0xFFFFFFFF)
    delay(500*us)
```
Unfortunately, it did not give any output on DAC. @jordens are we missing something?

gkasprow, 2020-09-30:
Actually, I observe 475...478mV DC on all DAC output pins.

jordens, 2020-10-01:
https://github.com/sinara-hw/Phaser/issues/116#issuecomment-699875812 ?

kaolpr, 2020-10-01:
Yes, I did comment out L63.

kaolpr, 2020-10-01:
And when I tried to use test mode the code never reached L63 either.

jordens, 2020-10-01:
What's "test mode"?
You said you can run the script to the end. If you can run the example.py script (artiq/phaser commits and bitstreams as in v0.4) to the end with that line commented out there should be dac output everywhere.

kaolpr, 2020-10-01:
> What's "test mode"?

I believe the mode that corresponds to `set_duc_cfg(select=1)`? Then `set_dac_test()` sets value that is output to DAC, am I right? At least that's what I tried when Greg reported that after `example.py` was finished with line 63 commented out (artiq/phaser commits as in v0.4 note) there was no output on DAC.

jordens, 2020-10-01:
It sets the value that's sent to the dac.

gkasprow, 2020-10-01:
Is there any kind of HW failure that can lead to such an issue?
Is the data integrity checked by the DAC somehow? There is a PARITY line, I assume DAC would set an alarm if something was wrong.
@kaolpr had issues with DAC initialization on this particular board, but the current code seems to work. The DAC does something, at least it sets the DC value correctly.

jordens, 2020-10-02:
I don't know.
`init()` runs lots of different tests, including several iotest. Parity check is enabled. So is every alarm that the datasheet describes. Some will turn of the DAC outputs if they trigger.
If the example script runs to the end (`dac_sleep` and `trf_ps` disabled) and does not indicate a relevant alarm by then, there should be output. Maybe check on a scope for transient output until the alarm triggers after the script is long done.
As mentioned [above](https://github.com/sinara-hw/Phaser/issues/116#issuecomment-699444342) the `sinara_tester` script will also create outputs.

hartytp, 2021-03-02:
@jordens Digging through the issues, I've lost track of what data you took. I see that you [measured](https://github.com/sinara-hw/Phaser/issues/101#issuecomment-682044877) a higher than expected current draw with the upconverters. Did you measure anything / anything unexpected with the baseband version?

hartytp, 2021-03-02:
IIRC the other suspicious observation was how hot the +-12V SMPS got given its light load. AFAICT this is consistent with the quiescent dissipation in that model and its poor thermal conductance to the PCB. This will be resolved in the next design iteration by moving to a more efficient (lower power) SMPS. cf https://github.com/sinara-hw/Phaser/issues/121

jordens, 2021-03-02:
I don't think I measured temperature and power on the baseband version.

hartytp, 2021-03-02:
This is data that @sotrirova took with a baseband phaser. I don't see anything suspicious here. 

[image]

hartytp, 2021-03-02:
hmmm...that's about 7W power consumption for the baseband unit. Assuming I've understood @gkasprow's power budget correctly the baseband consumption is 10W - something like 3W (plus converter losses) for the upconverter chip so about 7W. So this is a little higher than one might expect (given that the numbers in the table are generally worst-case), but not crazy.

[image]

hartytp, 2021-03-02:
anyway, it's hard to know precisely what to make of this. On reflection, I agree with @jordens comment that this does seem a bit high (e.g. I don't think that the FPGA is likely to be drawing max power). There isn't an obvious smoking gun in the thermal images @jordens posted (beyond the SMPS discussed above). 

It feels unlikely to be converter losses driving the extra dissipation: the are nominally only 1W so they'd have to be a long way off to make this kind of difference (at which point I'd expect to see a glowing inductor in the images).

Not clear what a good next step is here. Anyway, I acknowledge that there is something a little fisshy...

hartytp, 2021-03-02:
Having said that, it's also not clear that there *is* a problem here; the fact that Urukul came out significantly under is a product of the assumptions and methodology used there (not to mention how the relevant ICs were specified) and it's not clear that the same applies here. AFAICT there is nothing here that is inconsistent with the power budget and the extra losses that weren't accounted for (e.g. all that current passing through switching inductors).

Anyway, that doesn't feel 100% convincing, and I'd ideally like to know more but I don't see an obvious next step on that front.

gkasprow, 2021-03-03:
I want to replace the isolated converter with two high-efficiency buck and inverting converters used already in other designs.

gkasprow, 2021-03-11:
@hartytp can we do anything here?

hartytp, 2021-03-12:
@gkasprow I'm interested in your input here. I agree with @jordens that the power consumption is a bit high here (very close to the rated max of all chips unless I'm missing something) which could be a sign that something is wrong (or that we've messed up the power budget, but I don't think so). Maybe a converter with too much loss?

Anyway, I don't a simple next step here and I don't want this to be a release blocker so I suggest we leave the issue open for now with a plan to make further measurements once we have more hardware for testing

pathfinder49, 2021-04-09:
The Upconverter power consumption may related to the 5 V (>3.3 V) that are applied to the  VCC_LOx pins. https://github.com/sinara-hw/Phaser/issues/130#issuecomment-816768924

[3 more replies]

closed 2026-09-17 by kaolpr