sinara-hw/Phaser #117: Solder Mask and reflow reliability

jordens, 2020-09-19:
There are about a dozen instances where gaps between pads don't have solder mask or where components are really close together. Especially the decaps under the dac and the inductors (l31 etc) at the upconverters baseband inputs. The parts float around and move during reflow. Under the dac they merge and rotate because there is unmasked copper between parts. The inductors tip over and lean against each other (their placement is probably also bad in terms of coupling). If this isn't clear from the description I'll take a photo.
Let's fix that.

gkasprow, 2020-09-19:
OK, I know which ones you mean. we need to slightly separate the capacitors
[image]

jordens, 2020-09-21:
Yes. The tilting of the (blue) inductors (L31, L29, L33, L38) at the upconverter BB inputs can also be seen in your [photo](https://user-images.githubusercontent.com/4325054/75609757-83aad400-5b0b-11ea-8fe7-7b3fab54664c.jpg).
The pads look really small.

gkasprow, 2021-03-11:
fixed 
[image]

closed 2021-03-11 by gkasprow

gkasprow, 2021-03-11:
I changed the rule to 0.15mm instead of 75um

jordens, 2021-03-12:
Also the mixer inductors?
[image]

gkasprow, 2021-03-13:
I separated them and increased the pad size