sinara-hw/Phaser #118: Measure crosstalk

jordens, 2020-09-23:


hartytp, 2020-09-23:
This is RF0 out to RF1 out and vice versa, right?

jordens, 2020-09-23:
Yes

jordens, 2020-09-23:
Currently there is anywhere between 70 and 30 dB coupling loss between the two channels (at 1.2 GHz) because of the proximity to the panel RF shielding lip. On my board a light touch makes all the difference. Why can't we finally use longer SMA connectors everywhere that bridge the gap between the board and the panel without touching the panel and without having to extend the board while still fitting through the panel? I know that there are longer connectors.

gkasprow, 2021-03-11:
shall we use Amphenol 132203-16 for the RF path? they are 8mm longer.
Shall we move them by 8mm to keep a consistent connector length above the panel ?

gkasprow, 2021-03-13:
I found nice fitting connectors from Amp: 132203-14
[image]

closed 2021-03-13 by gkasprow