sinara-hw/Phaser #119: Optimize reconstruction filter

jordens, 2020-09-24:
For the upconverter variant we should closely follow the relevant eval boards (important for noise floor due to mixing products, bandwidth, flatness, proper matching, proper output power).

* Consider (or exclude) going DC coupled
* Consider the transformer network
* Double check and follow the guidelines and design procedure from the datasheets.

[image]
and others from [slua647a](https://www.ti.com/lit/ug/slua647a/slua647a.pdf)

[image]
from [TSW30H84EWM](https://www.ti.com/tool/TSW30H84EVM)

jordens, 2020-09-24:
https://github.com/sinara-hw/Phaser/issues/4#issuecomment-551014195

I think the approach may be good, we should just be sure what we want.

gkasprow, 2021-03-11:
@hartytp anything to do here?

hartytp, 2021-03-12:
I'm happy with the current filter. AFAICT we followed a sensible process to come up with the filter we chose so unless there is a specific issue with it I'm not aware of a need for change (but maybe @jordens has further thoughts?). We don't have a tight spec for what we're trying to achieve with this, which means that we can't optimize the solution too far.

jordens, 2021-03-12:
To close this, the current performance should be measured (power and frequency characteristics, crosstalk, IMD2 and IMD3 due to impedance errors). If that's ok, this can stay. IMO it would be better if someone does this thoroughly instead of later everybody for themselves in just their frequency window of interest.
AFAIK we don't even know the performance in a single window yet and whether it matches the design. I'd hesitate closing this.

jordens, 2021-03-12:
Yes. It's not a release blocker.

pathfinder49, 2021-04-13:
I've measured the filter attenuation for a range of DAC frequencies. The measurement was made by measuring the single ended power of a DAC output before and after the filter. I used the Keysight N2795A 1GHz, 1pF probe to reduce probe loading.

[image]

pathfinder49, 2021-04-13:
Note: The DAC output amplitude changes appreciably with frequency. The plot above does not represent the phaser output amplitude change with frequency.

hartytp, 2021-04-13:
Assuming I've followed this correctly, the comparison is with https://github.com/sinara-hw/Phaser/issues/4#issuecomment-551014195 This data looks really quite different to the simulation, so we should check that we have got this right...

pathfinder49, 2021-04-13:
@hartytp It's tricky to read of the plot you linked accurately. However, aside from the ~30 dB offset, my measurement looks consistent with the 10-300 MHz range from the simulation.

hartytp, 2021-04-13:
maybe you're right and I'm misreading it (I wouldn't worry about the 30dB offset until we've thought about units etc).

Can you compare to the [SPICE model](https://github.com/sinara-hw/Phaser/tree/master/Calc_SIM)

pathfinder49, 2021-04-13:
Seems like the Spice model is only for for the voltage regulator. The filter simulation requires Tina. I'll take a look at the trial later (unless @gkasprow can easily make a plot for comparison)

gkasprow, 2021-04-13:
[image]

gkasprow, 2021-04-13:
In the simulation, the peaking is 1.08dB

pathfinder49, 2021-04-13:
For context, I've measured the net power-frequency dependence for Phaser upconverter. There seems to be more significant variation than we have in the filter.

[image]
DUC in blue, NCO in orange

The DUC drop-off at higher frequencies is expected.

jordens, 2022-09-12:
If we could improve the DAC AA filter (relevant for both variants) and the DAC-Mixer matching (Upvonverter), that would be really useful. Right now with the AA filter doing -3 dB already before 300 MHz and the overall match having 6 dB ripple over 250 MHz, it's certainly below what's possible.
Something like -0.2 dB @ 350 MHz, -35 dB at 650 MHz should be possible. And the impedances thoroughly reviewed.

[image]
[image]
https://rf-tools.com/lc-filter/

jordens, 2022-09-12:
Alternatively LTCC filters, https://www.minicircuits.com/pdfs/LFCN-320.pdf (100 Ohm as a balanced pair)

gkasprow, 2022-09-13:
@jordens It didn't go for production yet so we will update it.
[image]

gkasprow, 2022-09-13:
it fits
[image]

jordens, 2022-09-13:
For baseband, if we do 200 R balanced, then the traces should also have that impedance up to the balun.

If possible, I'd separate the P and L inductors a bit to minimize coupling.

Interestingly, on the eval board they put the source termination after the AA filter. Not sure if that's worth it.
[image]

In the upconverter variant we need to also then make the load impedance 200 R after the filter. By using a 2:1 balbal and the datasheet 100 R termination at the mixer input (TC2-1T+ in the footprint of the baseband 4:1 balun).

[image]

Alternatively a 200R termination and no balbal. But I'd prefer the balbal and 100 R termination.

gkasprow, 2022-09-13:
@jordens we cannot change the trace impedance between assembly options.
 but traces are so short that it doesn't really matter.
For the moment in upconverter variant we use 50Ohm DAC load and 2x60R4 series resistors at the output of the filter which together with 91Ohm load make more or less 200Ohm

gkasprow, 2022-09-13:
200R impedance is impossible with current PCB stackup. Even two 80um traces separated by 1mm would give 130Ohm

jordens, 2022-09-13:
It was specifically the Upconverter variant that had large ripples (https://github.com/sinara-hw/Phaser/issues/119#issuecomment-818931327) much larger than the simulation/design (https://github.com/sinara-hw/Phaser/issues/4#issuecomment-551014195).
Isn't the match between the 50 R at the DAC and the 200 R at the TRF a bit bad? Both in simulation and in practice?
Your original plan from the datasheet (https://github.com/sinara-hw/Phaser/issues/4#issuecomment-550489607) had 250 R on both ends.

jordens, 2022-09-13:
Ok. Then let's stick with 100 R balanced trace impedance throughout. I'm also not too worried about that.

gkasprow, 2022-09-13:
what we can do is to use 100R DAC loads in both assembly variants

jordens, 2022-09-13:
@pathfinder49 @FabianSchwartau what do you think?

gkasprow, 2022-09-13:
the XFMR_MINI-CIRCUITS TC2-72T+ has same footprint

gkasprow, 2022-09-13:
the schematic for upconverter variant would be as follows. I added caps at the output
[image]

jordens, 2022-09-13:
The 100 R at the DAC needs to go (everything should be the same up to the balbal. Otherwise looks good. Also TC2-72T+ in favor of TC2-1T+ for better balance and flatter response.
And in both cases there can't be a series C in the filter (remove C135 C138) as the DC current path to the center tap needs to be there.

gkasprow, 2022-09-13:
fixed

closed 2022-09-13 by gkasprow

jordens, 2022-09-13:
@gkasprow As mentioned above, I'd prefer the straight 100 R differential termination as close as possible to the modulator as in the eval board. Can we 0R those inductors?

[image]

gkasprow, 2022-09-14:
OK

pathfinder49, 2022-09-14:
> @pathfinder49 @FabianSchwartau what do you think?

I haven't thought too much about different filter options. However, I'm generally in favour of anything increasing Phaser output power. The power is painfully low and in some scenarios the signal to noise ratio was limiting for me with either variant.

jordens, 2022-09-14:
@pathfinder49 For that to be the case it would need to be attenuated so low that the DAC(+Upconverter) noise gets near the thermal floor, something like -30 dBm output for typical frequencies. It would surprise me if that's the case (it wasn't in my tests). Could you check where the power is lost, what should be achieved, and propose fixes?

gkasprow, 2022-09-14:
We can think of adding high linearity power stage similar to Booster HL but lower power.

pathfinder49, 2022-09-14:
> @pathfinder49 For that to be the case it would need to be attenuated so low that the DAC(+Upconverter) noise gets near the thermal floor, something like -30 dBm output for typical frequencies. It [...]

In my case I produce a ~ +40 dBm, 3 GHz signal which is too high a power to isolate the amplifiers using most fast switches. In my setup, the noise from the amplifiers with enough gain to boost the signal to +40 dBm drives qubit transitions on the <100 ms timescale when using Phaser Upconverter. The Phaser noise makes this even worse, but I can extinguish that with an inline switch as it isn't as high power. Of cause, this is a somewhat unusual setup (near field microwave gates). This dominates the decoherence time of my qubits. In my case moving from Phaser Upconverter to Phaser Baseband with external IQ upconversion gave ~10 dB increased output power, necessitating less amplifier gain and increasing coherence times. Of cause, there are ways around this by designing an amplifier chain with this in mind. However it also wouldn't be such an issue if realistic Phaser-upconverter signals weren't <-20 dBm.

There are also other signal-to-noise issues that should benefit, like the leakage of the upconverter LO and of the reference clock harmonics.

pathfinder49, 2022-09-14:
> We can think of adding high linearity power stage similar to Booster HL but lower power.

I'm not sure I'd advocate for an on-phaser gain stage. Phaser is already fairly noisy, especially in the upconverter variant. Therefore, I wouldn't change the design too much for high signal-to-noise applications. (I guess this depends on what the community wants). However, given the limited power available, we shouldn't throw away the little power we have as-is (especially in the up-converter variant).

jordens, 2022-09-14:
Your final power and the overall amplification have nothing to do with the SNR as long as yo aren't limited by the thermal noise floor. Did you distinguish the noise floor hitting thermal floor from the effect of close-to-LO noise (1/f etc)? This is the chance to look at it and figure out whether what you claim is compatible with a nominal -3 dB mixer gain, the noise of the upconverter and the DAC, and the carrier power vs thermal.

jordens, 2022-09-14:
I don't see the need or benefit from a on-phaser gain stage. If there is a SNR problem, it's not the solution.

hartytp, 2022-09-14:
> In my case I produce a ~ +40 dBm, 3 GHz signal which is too high a power to isolate the amplifiers using most fast switches. In my setup, the noise from the amplifiers with enough gain to boost the [...]

I don't follow this I'm afraid. It sounds like in the situation you're describing you're getting decoherence just from the amplifier chain, even with no phaser (amplifier input terminated) because you're using very high gain amplifiers. Is that correct? If so, this doesn't sound like an issue with phaser but rather with the amplifier chain (as you say, the issue can be resolved by designing the amplifier chain differently).

hartytp, 2022-09-14:
Do we really get -20dBm out of the phaser upconverter? My understanding is that we expect ~0dBm out of the DAC and a few dB of conversion loss in the upconverter so something like -6dBm including filters. Are my expectations wrong or is there a bug in the design?

pathfinder49, 2022-09-14:
> Your final power and the overall amplification have nothing to do with the SNR as long as yo aren't limited by the thermal noise floor. Did you distinguish the noise floor hitting thermal floor from [...]

I distinguished the effects by measuring coherence time response to placing attenuators at diffrerent points in the microwave signal chain. When not attenuated or switched the Phaser upconverter output noise dominates over this effect. However, it is straightforward to include a TTL switch in the unconverted phaser output (and I did).

> Do we really get -20dBm out of the phaser upconverter? My understanding is that we expect ~0dBm out of the DAC and a few dB of conversion loss in the upconverter so something like -6dBm including filters. Are my expectations wrong or is there a bug in the design?

The absolute maximum I can get around 3 GHz [is -10 dBm](https://github.com/sinara-hw/Phaser/issues/130#issuecomment-817925371) with a breakdown [here](https://github.com/sinara-hw/Phaser/issues/130#issuecomment-818631631).

By the time you're [not sitting on the filter resonance peak](https://github.com/sinara-hw/Phaser/issues/130#issuecomment-818635081), avoid digital clipping, and use two tones you lose another 10 dB. So in realistic use cases the available power per tone is -20 dBm (plus cable losses).

jordens, 2022-09-15:
Still unclear what other settings you had there. I [asked](https://github.com/sinara-hw/Phaser/issues/130#issuecomment-817988851) for that info but there was no response. Please review the filter, identify why it doesn't meet the goals, and propose solutions if you are really interested in this. E.g. check whether the proposed would work, or even a 4:1 balbal with a 200 R termination at the mixer as proposed. Doing a quick test on hardware is also helpful.

Also it's not clear to me why an external mixer with the baseband variant would help with power over the upconverter. It uses the exact same reconstruction filter with the same loss and the mixer will have similar gain. Also with the I and Q outputs now being significantly different circuits matching group delays, amplitudes, phases, and DC offsets would be significantly more annoying, more so if this is supposed to work over a non-trivial frequency range.

One thing that may be doable is adding an RF switch to kill the output, assuming we can find a good broad band one.
But then again it's easy to just add 32 dB from the attenuator while not doing gates. That's much better broad band noise suppression than the 5 dB you are hunting.
Also if the DAC is outputting DC, I'm pretty sure broadband noise at the upconverter input will be below the datasheet value of the DAC output for non-DC. Another easy knob to use.

pathfinder49, 2022-09-15:
> Please review the filter, identify why it doesn't meet the goals, and propose solutions if you are really interested in this. E.g. check whether the proposed would work, or even a 4:1 balbal with a 200 R termination at the mixer as proposed. Doing a quick test on hardware is also helpful.

I currently have no bandwidth for any further work on Phaser. 

> Also it's not clear to me why an external mixer with the baseband variant would help with power over the upconverter.

Maybe take a look at where the power losses occur?

> That's much better broad band noise suppression than the 5 dB you are hunting

I'm not hunting anything. You asked me for my thoughts, which I have given. I have what I need and you are welcome to do any further characterisation you need.

jordens, 2022-09-15:
Good luck with that attitude.