sinara-hw/Phaser #120: DAC output termination resistors far away from DAC

jordens, 2020-09-24:
Should and can be much closer. Required and stressed multiple times in the datasheets. (The impedance between the DAC output and the resistors does not match the trace dimensions).

hartytp, 2020-09-24:
Thanks @jordens 

One thing I was curious about here is why we didn't follow the EVM for the DAC outputs.

[image]

[image]

jordens, 2020-09-24:
This is more for #110 
It depends on the variant and on the impedance at the balun and also the series resistors before the balun.

gkasprow, 2021-03-13:
Fixed
[image]

closed 2021-03-13 by gkasprow